m-labs/artiq #218: Feature request: TTL input read current value

dleibrandt, 2016-01-07:
I think the ability to read the current value of a TTL input (i.e., high or low, even if there are no rising or falling edges) would be useful.

Example use case:
Our experiments often include several (hardware) PID controllers which are used for things like locking laser frequencies.  These controllers often have a TTL output that indicates if the system is currently locked.  I would like to feed all of these monitor TTLs into my ARTIQ FPGA, and check that they are all locked at before starting to run an experiment.  (At the end of the experiment, I will also check that they didn't come unlocked during the experiment, but the ability to do that is already present.)

jordens, 2016-09-03:
The purpose of this addition is to enable the readout of the current value of a TTL input in the absence of TTL transitions. The gateware, runtime, and driver shall be updated to support the following user functions:
A sample() function, that instructs the RTIO core to read the value of the TTL input at the current value of ‘now’ (the RTIO counter).
A get_value() function, that returns the value of the sample previous triggered using sample()
A get_value_nonrt() function, that samples, returns the value, and calls core.break_realtime(). This is a convenience function for use in situations without strict real-time constraints.

sbourdeauducq, 2016-09-07:
In addition to the specified function I have added a "watch" API that does what you describe in the example without race conditions.
https://m-labs.hk/artiq/manual-master/core_drivers_reference.html#artiq.coredevice.ttl.TTLInOut.watch_done

closed 2016-09-07 by sbourdeauducq