m-labs/artiq #219: Core device storage

whitequark, 2016-01-08:
We need persistent storage in RAM of coredevice. Lists of integers up to 10M should be stored. Storage and retrieval is performed using string keys. Retrieval should not copy lists.

whitequark, 2016-01-08:
API: `self.core.get_cache`/`self.core.set_cache`. On error return empty list. If a list was returned by `get_cache` then `set_cache` will refuse to set it until the current experiment terminates, to maintain memory safety.

whitequark, 2016-01-08:
@sbourdeauducq Question: where should we stick the heap for the cache and how large should it be?

whitequark, 2016-01-08:
Nevermind "where", lwip does not give much choice in this regard. But "how large" still stands.

jboulder, 2016-01-12:
Please add to documentation.

jboulder, 2016-01-12:
Would also be good to benchmark performance. 
* from Master PC to DRAM
* from DRAM to Core Device

whitequark, 2016-01-12:
There is no DRAM to Core Device step: any data stored in the cache is used directly when you extract it.

whitequark, 2016-01-16:
Documentation added.

commit 9366a294 (2016-01-10): Implement core device storage (fixes #219).
closed 2016-01-20 by sbourdeauducq