m-labs/artiq #220: rewrite/replace the pyqtgraph docking system

sbourdeauducq, 2016-01-10:
There are a number of problems with it:
* the look-and-feel is mediocre. Double-click to detach tab is not very intuitive. Order of tabs cannot be changed. Compare with the tabs in Google Chrome or Sublime Text.
* it is buggy:
  * https://github.com/m-labs/artiq/commit/2199eadb9962811f19f0c78891c62d7c5a246fe2#diff-523ba5dbaffb3068cb299aaa89ab0456R215
  * it reproducibly explodes and makes the application unusable if the dock saved state references a non-existent dock
  * it randomly corrupts the dock save/restore state (#104, #158)
* pyqtgraph is generally hacky, e.g. it meddles with the PyQt namespaces (). Removing the main GUI's dependency on pyqtgraph would contain the mess to those applets that use it.

whitequark, 2016-01-10:
https://doc.qt.io/qt-4.8/qdockwidget.html ?

sbourdeauducq, 2016-01-15:
The pyqtgraph dock says "Docking system that replaces/complements Qt’s dock system to allow more complex (and more predictable) docking arrangements". Not sure how much of that is true/actually provides something useful.

sbourdeauducq, 2016-01-15:
When we have a decent docking system, we should also bring the corresponding experiment dock into focus when it is already open and the user double-clicks on the experiment in the explorer.

sbourdeauducq, 2016-02-13:
https://woboq.com/blog/qdockwidget-changes-in-56.html seems to have most of the features of pyqtgraph docks, plus a few extras (e.g. tab reordering). It will be part of Qt 5.6 (release due March 3rd) and the company developing it offers commercial support.

sbourdeauducq, 2016-02-14:
Replaced with QDockWidget.

closed 2016-02-14 by sbourdeauducq