m-labs/artiq #221: package pyqt5

sbourdeauducq, 2016-01-10:
Needed for window embedding features, and for fixing bugs inside Qt.

Anaconda only offers pyqt4 in the main repository; but there is this: http://stackoverflow.com/questions/25468397/using-anaconda-python-with-pyqt5

whitequark, 2016-02-14:
Done.

sbourdeauducq, 2016-02-17:
As discussed:
 * use host libxcb library (keyboard does not work at all on Linux)
 * qt4 + qt5 coexistence is needed (otherwise conflicts with many anaconda packages)

whitequark, 2016-02-22:
This is now packaged, uses host libxkbcommon and coexists with Qt4.

closed 2016-02-22 by whitequark

sbourdeauducq, 2016-02-22:
I tested it (with the single-command installation in the doc) on Linux on fresh Anaconda and on fresh Miniconda (since both are mentioned in the doc, we must make sure that it works with both). Have you done the same on Windows?

whitequark, 2016-02-22:
I verified that it works in a fresh 64-bit Anaconda and Miniconda. However, only in Anaconda, the following message is displayed when closing artiq_gui:
[image]

whitequark, 2016-02-22:
Ah nevermind, looks like I had a stale `artiq_gui.pyon`.