m-labs/artiq #224: returning tuple from kernel causes AssertionError

sbourdeauducq, 2016-01-13:
```python
	@kernel
	def test_return(self):
		return 1, 2

	def run(self):
		print(self.test_return())
```

```
Traceback (most recent call last):
  File "/home/sb/artiq/artiq/frontend/artiq_run.py", line 154, in <module>
    main()
  File "/home/sb/artiq/artiq/frontend/artiq_run.py", line 150, in main
    return run(with_file=True)
  File "/home/sb/artiq/artiq/frontend/artiq_run.py", line 136, in run
    raise exn
  File "/home/sb/artiq/artiq/frontend/artiq_run.py", line 129, in run
    exp_inst.run()
  File "repository/nointerleave.py", line 27, in run
    print(self.test_return())
  File "/home/sb/artiq/artiq/language/core.py", line 197, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/sb/artiq/artiq/coredevice/core.py", line 95, in run
    object_map, kernel_library, symbolizer = self.compile(function, args, kwargs, set_result)
  File "/home/sb/artiq/artiq/coredevice/core.py", line 81, in compile
    library = target.compile_and_link([module])
  File "/home/sb/artiq/artiq/compiler/targets.py", line 141, in compile_and_link
    return self.link([self.compile(module) for module in modules],
  File "/home/sb/artiq/artiq/compiler/targets.py", line 141, in <listcomp>
    return self.link([self.compile(module) for module in modules],
  File "/home/sb/artiq/artiq/compiler/targets.py", line 83, in compile
    llmod = module.build_llvm_ir(self)
  File "/home/sb/artiq/artiq/compiler/module.py", line 84, in build_llvm_ir
    return llvm_ir_generator.process(self.artiq_ir, attribute_writeback=True)
  File "/home/sb/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 413, in process
    self.process_function(func)
  File "/home/sb/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 555, in process_function
    llinsn = getattr(self, "process_" + type(insn).__name__)(insn)
  File "/home/sb/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 925, in process_Closure
    llvalue = self.llbuilder.insert_value(llvalue, self.map(insn.target_function), 1,
  File "/home/sb/artiq/artiq/compiler/transforms/llvm_ir_generator.py", line 405, in map
    llfun.args[0].add_attribute('sret')
  File "/usr/local/lib/python3.5/dist-packages/llvmlite_artiq/ir/values.py", line 476, in add_attribute
    self.attributes.add(attr)
  File "/usr/local/lib/python3.5/dist-packages/llvmlite_artiq/ir/values.py", line 336, in add
    assert name in self._known
AssertionError
```

whitequark, 2016-01-13:
That has nothing to do with tuples, you need to update llvmlite.

sbourdeauducq, 2016-01-13:
Well, the way to reproduce that problem is to return a tuple. It does not manifest itself when returning a scalar. Anyway, updating llvmlite solved it.

closed 2016-01-13 by sbourdeauducq

whitequark, 2016-01-13:
Returning any complex structure by value, e.g. lists, would do it.

sbourdeauducq, 2016-01-13:
No, lists do this:
```
repository/nointerleave.py:24:10-24:16: error: cannot return a mutable value that does not live forever
		return [2, 3]
		       ^^^^^^
repository/nointerleave.py:22:2: note: the value of the expression is alive from this point...
	@kernel
	^      
repository/nointerleave.py:24:16: note: ... to this point
		return [2, 3]
		             ^
repository/nointerleave.py:24:10-24:16: note: this expression has type list(elt=int(width=32))
		return [2, 3]
		       ^^^^^^
```

whitequark, 2016-01-14:
Try a list from the core device storage...