m-labs/artiq #227: compiler crashes

jboulder, 2016-01-15:
This is a compiler bug says SB. 

Also the colors are not right in the GUI. The color escape sequences were still displayed in the GUI Log. 

```
INFO:worker(10490):print:<synthesized>:1:1-1:62: error: host object does not have an attribute 'counts'
INFO:worker(10490):print:`<repository.std_include.StdDetect object at 0x7fd22b65d550>`
INFO:worker(10490):print:^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
INFO:worker(10490):print:/home/rabi/gitlab/nistpenning/artiq/repository/std_include.py:74:9-74:20: note: attribute accessed here
INFO:worker(10490):print:        self.counts[self.rep_i] = cts
INFO:worker(10490):print:        ^^^^^^^^^^^                  
INFO:worker(10490):print:Traceback (most recent call last):
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/coredevice/core.py", line 76, in compile
INFO:worker(10490):print:    stitcher.finalize()
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/compiler/embedding.py", line 476, in finalize
INFO:worker(10490):print:    inferencer.visit(self.typedtree)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in visit
INFO:worker(10490):print:    return [self.visit(elt) for elt in obj]
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
INFO:worker(10490):print:    return [self.visit(elt) for elt in obj]
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
INFO:worker(10490):print:    return self._visit_one(obj)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
INFO:worker(10490):print:    return getattr(self, visit_attr)(node)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/compiler/transforms/inferencer.py", line 1150, in visit_FunctionDefT
INFO:worker(10490):print:    self.generic_visit(node)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
INFO:worker(10490):print:    self.visit(getattr(node, field_name))
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in visit
INFO:worker(10490):print:    return [self.visit(elt) for elt in obj]
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
INFO:worker(10490):print:    return [self.visit(elt) for elt in obj]
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
INFO:worker(10490):print:    return self._visit_one(obj)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
INFO:worker(10490):print:    return getattr(self, visit_attr)(node)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/compiler/transforms/inferencer.py", line 901, in visit_Assign
INFO:worker(10490):print:    self.generic_visit(node)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
INFO:worker(10490):print:    self.visit(getattr(node, field_name))
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in visit
INFO:worker(10490):print:    return [self.visit(elt) for elt in obj]
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
INFO:worker(10490):print:    return [self.visit(elt) for elt in obj]
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
INFO:worker(10490):print:    return self._visit_one(obj)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
INFO:worker(10490):print:    return getattr(self, visit_attr)(node)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/compiler/transforms/inferencer.py", line 219, in visit_SubscriptT
INFO:worker(10490):print:    self.generic_visit(node)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
INFO:worker(10490):print:    self.visit(getattr(node, field_name))
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
INFO:worker(10490):print:    return self._visit_one(obj)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
INFO:worker(10490):print:    return getattr(self, visit_attr)(node)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/compiler/transforms/inferencer.py", line 93, in visit_AttributeT
INFO:worker(10490):print:    loc=node.loc)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/compiler/embedding.py", line 330, in _unify_attribute
INFO:worker(10490):print:    self.engine.process(diag)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/pythonparser/diagnostic.py", line 157, in process
INFO:worker(10490):print:    raise Error(diagnostic)
INFO:worker(10490):print:pythonparser.diagnostic.Error: <synthesized>:1:1-1:62: error: host object does not have an attribute 'counts'
INFO:worker(10490):print:`<repository.std_include.StdDetect object at 0x7fd22b65d550>`
INFO:worker(10490):print:^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
INFO:worker(10490):print:/home/rabi/gitlab/nistpenning/artiq/repository/std_include.py:74:9-74:20: note: attribute accessed here
INFO:worker(10490):print:        self.counts[self.rep_i] = cts
INFO:worker(10490):print:        ^^^^^^^^^^^                  
INFO:worker(10490):print:
INFO:worker(10490):print:The above exception was the direct cause of the following exception:
[... 9 lines cut ...]
INFO:worker(10490):print:    object_map, kernel_library, symbolizer = self.compile(function, args, kwargs, set_result)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/coredevice/core.py", line 87, in compile
INFO:worker(10490):print:    raise CompileError(error.diagnostic) from error
INFO:worker(10490):print:artiq.coredevice.core.CompileError: 
INFO:worker(10490):print:<synthesized>:1:1-1:62: error: host object does not have an attribute 'counts'
INFO:worker(10490):print:`<repository.std_include.StdDetect object at 0x7fd22b65d550>`
INFO:worker(10490):print:^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
INFO:worker(10490):print:/home/rabi/gitlab/nistpenning/artiq/repository/std_include.py:74:9-74:20: note: attribute accessed here
INFO:worker(10490):print:        self.counts[self.rep_i] = cts
INFO:worker(10490):print:        ^^^^^^^^^^^                  
INFO:worker(10490):print:
INFO:worker(10490):print:During handling of the above exception, another exception occurred:
INFO:worker(10490):print:
INFO:worker(10490):print:Traceback (most recent call last):
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/runpy.py", line 170, in _run_module_as_main
INFO:worker(10490):print:    "__main__", mod_spec)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/runpy.py", line 85, in _run_code
INFO:worker(10490):print:    exec(code, run_globals)
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 250, in <module>
INFO:worker(10490):print:    main()
INFO:worker(10490):print:  File "/home/rabi/anaconda3/envs/newartiq/lib/python3.5/site-packages/artiq/master/worker_impl.py", line 234, in main
INFO:worker(10490):print:    message = "Cannot compile {}\n".format(experiment_file) + exc.render_string()
INFO:worker(10490):print:AttributeError: 'CompileError' object has no attribute 'render_string'

```

whitequark, 2016-01-15:
I cannot do anything without the source of `repository/std_include.py`.

As for the error itself, yes, it is horribly mangled. I will fix that shortly.

jboulder, 2016-01-15:
```python
from artiq import *
import numpy as np
from scipy.optimize import curve_fit
import time
import logging

class StdCool(HasEnvironment):
    """Standard cooling and state perparation
    """
    def build(self):
        self.core = self.get_device("core")
        self.doppler_sw = self.get_device("doppler_sw")
        self.repump_sw = self.get_device("repump_sw")
        self.led = self.get_device("led")

        gui_group = "Standard Cool"
        self.cool_t = self.get_argument("cool_t",
                           NumberValue(default=3e-3, unit="ms", scale=1e-3,
                                       step=1e-3, ndecimals=0,
                                       min=100e-6, max=1), gui_group)
        self.repump_t = self.get_argument("repump_t",
                           NumberValue(1e-3, unit="ms", scale=1e-3, step=1e-3,
                                       ndecimals=0, min=0, max=1), gui_group)

    @kernel
    def go(self):
        """cooling and repumping
        """
        self.led.on()
        self.doppler_sw.on()
        self.repump_sw.on()
        delay(self.cool_t)
        self.doppler_sw.off()
        delay(self.repump_t)
        self.repump_sw.off()
        self.led.off()

class StdDetect(HasEnvironment):
    """state detection
    """
    def build(self):
        self.core = self.get_device("core")
        self.doppler_sw = self.get_device("doppler_sw")
        self.repump_sw = self.get_device("repump_sw")
        self.led = self.get_device("led")
        self.sideview_pmt = self.get_device("sideview_pmt")

        gui_group = "Standard Detect"
        self.hist_nbins = self.get_argument("hist_nbins",
                           NumberValue(60, step=10, ndecimals=0, min=0),
                           gui_group)
        self.hist_nmax = self.get_argument("hist_nmax",
                           NumberValue(200, step=10, ndecimals=0,
                                       min=10), gui_group)
        self.nreps = int(self.get_argument("nreps",
                           NumberValue(100, step=10, ndecimals=0, min=2,
                                       max=1000), gui_group))
        self.detect_t = self.get_argument("detect_t",
                           NumberValue(1e-3, unit="ms", scale=1e-3, step=1e-4,
                                       ndecimals=1, min=0, max=1), gui_group)
        self.counts = [0]*self.nreps
        self.rep_i = 0

    @kernel
    def rep(self):
        """measurement; single experiment repetition
        """
        self.led.on()
        self.doppler_sw.on()
        self.sideview_pmt._set_sensitivity(1)
        delay(self.detect_t)
        self.led.off()
        self.doppler_sw.off()
        self.sideview_pmt._set_sensitivity(0)
        cts = self.sideview_pmt.count()
        self.counts[self.rep_i] = cts
        self.rep_i += 1

    @kernel
    def rep_start(self):
        """initialize count buffer
        """
        self.rep_i = 0
        self.counts = [0 for _ in range(int(self.nreps))]

    def hist_rpc(self, counts):
        """update histogram in GUI
        """
        hist, hist_bins = np.histogram(counts, self.hist_nbins,
            range=(0, self.hist_nmax))
        # update histogram
        self.set_dataset("common_hist1_bins", hist_bins,
                         broadcast=True, persist=True, save=False)
        self.set_dataset("common_hist1_counts", hist,
                         broadcast=True, persist=True, save=False)
        self.set_dataset("detect_bright_mean", float(np.mean(counts)),
                         broadcast=True, persist=True, save=False)
        # update timeseries

        def stripchart_roll(xs, x0):
            xsnp = np.roll(np.array(xs), -1)
            xsnp[-1] = float(x0)
            return xsnp
        # I only want to run these one to initialize the dataset.
        # They could reside in program file that I only run as needed to init things.
        # For now I'll just comment them out after the first run.
        current_bright_mean = self.get_dataset('detect_bright_mean',
                                               default=100)
        x = self.get_dataset("default_bright_mean_strip_x",
                             default=[current_bright_mean]*100)
        y = self.get_dataset("default_bright_mean_strip_y",
                             default=[int(time.time())]*100)
        y = stripchart_roll(y, np.mean(counts))
        x = stripchart_roll(x, int(time.time()))
        self.set_dataset("default_bright_mean_strip_x", x,
                         broadcast=True, persist=True)
        self.set_dataset("default_bright_mean_strip_y", y,
                         broadcast=True, persist=True)

    def hist_model(n, mean):
        return mean**n*np.exp(-1*mean)/np.math.factorial(n)

    def sim_nreps(self):
        # add small random walk to mean for dramatic effect
        bright_mean = self.get_dataset('detect_bright_mean', default=100)
        mean = bright_mean + np.random.normal(loc=0, scale=3)

        return np.random.poisson(abs(mean), size=self.nreps)
```

Create the error by removing the ``self.counts`` and ``self.rep_i`` initialization from ``build()``

Note two things that the old compiler did better:
* no requirement for this initialization (e.g. for attributes that reside entirely on device).
* interleaving of ``sideview_pmt.gate_rising``, which had to be done manually in this code.

jboulder, 2016-01-15:
Also, setting nreps to 5 causes ``artiq.coredevice.exceptions.InternalError: Hardware exception 6 at PC 0x42020b08, EA 0x00000023``

whitequark, 2016-01-15:
> no requirement for this initialization (e.g. for attributes that reside entirely on device)

Initialization is intentionally required; this is a feature, not a bug. This way, the typechecker will accept accesses to attributes, regardless of what kernel code is pulled in by any particular kernel that you call. Otherwise, the typechecker would exhibit "action at a distance": adding or removing a piece of code will affect whether another seemingly unrelated piece of code compiles. The old compiler was able to accept this usage at the cost of being less rigorous almost everywhere else.

> interleaving of sideview_pmt.gate_rising, which had to be done manually in this code.

Can you post the `with parallel:` statement you were trying to write?

whitequark, 2016-01-15:
> Also, setting nreps to 5 causes artiq.coredevice.exceptions.InternalError: Hardware exception 6 at PC 0x42020b08, EA 0x00000023

This is certainly a bug and I will address that soon.

jboulder, 2016-01-15:
```python
with parallel:
        self.led.pulse(self.detect_t)
        self.doppler_sw.pulse(self.detect_t)
        self.sideview_pmt.gate_rising(self.detect_t)
```

whitequark, 2016-01-15:
Ah yes, the same issue as (somewhat misleadingly named) #193; see https://github.com/m-labs/artiq/issues/193#issuecomment-165700170 for an outline of my proposed solution.

jboulder, 2016-01-16:
default_std.py

```python
from artiq import *
from artiq.coredevice.exceptions import RTIOUnderflow
import numpy as np
import time
import repository.std_include as std_include
import logging

def print_underflow():
    print("Joe's RTIO underflow occured")

class DefaultStd(EnvExperiment):
    """default_std

    Default experiment that runs at priority 0. Cooling, histogram.
    """

    def build(self):
        self.core = self.get_device("core")
        self.scheduler = self.get_device("scheduler")
        self.cool = std_include.StdCool(parent=self)
        self.no_cool_t = self.get_argument("no_cool_t",
                           NumberValue(default=10e-3, unit="ms", scale=1e-3,
                                       min=0, max=200e-3,
                                       ndecimals=0, step=5e-3))
        self.sim = self.get_argument("simulate ions",
                                     BooleanValue(default=True))
        self.detect = std_include.StdDetect(parent=self)

    @kernel
    def run_kernel(self):
        self.detect.rep_start()
        for i in range(int(self.detect.nreps)):
            self.core.break_realtime()
            self.cool.go()
            # experiment meat would go here if this were a fancy experiment
            self.detect.rep()
            delay(self.no_cool_t)
        self.detect.hist_rpc(self.detect.counts)

    def run_sim(self):
        counts = self.detect.sim_nreps()
        self.detect.hist_rpc(counts)
        t_one_rep = self.cool.cool_t + self.cool.repump_t + \
                    self.no_cool_t + self.detect.detect_t
        time.sleep(t_one_rep*self.detect.nreps)

    def run(self):
        try:
            if self.sim:
                while True:
                    self.run_sim()
                    self.scheduler.pause()
            else:
                while True:
                    self.run_kernel()
                    self.core.comm.close()
                    self.scheduler.pause()

        except TerminationRequested:
            logging.info("Terminated gracefully")
```

whitequark, 2016-01-16:
GUI was fixed. The issue which caused InternalError was triaged as #232. The solution is to replace

    self.counts = [0 for _ in range(int(self.nreps))]

with

    self.counts[:] = [0 for _ in range(int(self.nreps))]

The former code should have been rejected with a diagnostic, but that did not happen, causing a memory safety bug.

Can you confirm that the solution above works for you?

whitequark, 2016-01-16:
BTW there is no need to write `range(int(self.nreps))`, just `range(self.nreps)` is enough.

sbourdeauducq, 2016-01-16:
Yes, the problem was ``self.nreps`` was a float in Joe's initial code; I converted it once and for all in ``build`` but had not removed all the ad-hoc conversions yet when that code was pasted.

closed 2016-01-20 by sbourdeauducq