m-labs/artiq #230: features needed for experiment browser/replay

sbourdeauducq, 2016-01-15:
* dataset checkpoints
* ability to write a visual representation of the results to a PNG picture (a function to get a path and prefix so the PNG is close to the HDF5 output)
* ability to open an experiment dock from a HDF5 file with:
  * repository revision preloaded to the HDF5 value
  * arguments preloaded to the HDF5 values
  * reanalyze button in addition to submit
* (optional) the file selection dialog for opening the dock from HDF5 can display the PNG

sbourdeauducq, 2016-04-04:
To avoid complexities with overlays/checkpoints on the dataset-DB and master interaction, browser/replay will be implemented as a separate application. It will be entirely standalone  (not connecting to the master) and run re-analyze locally. This also allows users to run analyses offline (when disconnected from the rest of the ARTIQ system) with just the HDF5 and the experiment source file(s).

jordens, 2016-05-25:
All features should be there. Might benefit from caching the h5 opening/h5 data or having a common h5 data provider. Icon provider, metadata tooltip provider, dataset loader, argument loader each open the h5 file.

closed 2016-05-25 by jordens

jboulder, 2016-06-13:
It took me a while to figure out how this feature. Here's the steps. 
1. Launch browser ```$ artiq_browser```
2. From Experiment menu Open an experiment file (.py). An experiment control widget appears. 
3. On left side of screen is file hierarchy viewer labelled Files. Navigate to results directory and locate a .h5 for the experiment type you opened in step 2. Drag the .h5 file on top of the experiment control widget.
4. Click Analyze in the widget. 
5. Re-create the applets needed to visualize your datasets.
6. Subsequent .h5 files of the same type can be visualized by dragging and clicking Analyze. 

Without some guidance users may find that setup is non-intuitive. I was completely missing that loading .py is a necessary step. 

It would be convenient if as different .h5 are selected...
* the corresponding .py  automatically loaded (based on git hash?)
* Analyze were automatically performed
* Applets are reconstituted as when the user originally ran the experiment. I know this breaks the present strict separation of GUI and Master roles. 

Thanks for implementing this!