m-labs/artiq #234: rid scalability/uniqueness/sortability

jordens, 2016-01-17:
Scanning the entire output data filesystem does not scale. There are usually millions of files there (I checked).
If run-ids do not need to be sortable, they don't need to be monotonically increasing and they could be `uuids`. But sortability is nice for browsing the filesystem, so let's keep it.
Can't we just put the last/current run id into e.g. the dataset db?

https://github.com/m-labs/artiq/blob/master/artiq/master/worker_db.py#L90

sbourdeauducq, 2016-02-13:
I would use a dedicated file for that and rewrite it at each RID assignment.

jordens, 2016-02-15:
You will want to keep the current code to a) reconstruct that pointer file in case it gets lost and b) tarnsition old repos.

commit 72a993af (2016-02-15): master: cache last RID. Closes #234
closed 2016-02-15 by sbourdeauducq