m-labs/artiq #236: test_loopback_count() needs rewriting for new parallel constraints

jordens, 2016-01-18:
```python
======================================================================
ERROR: test_loopback_count (artiq.test.coredevice.rtio.CoredeviceTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/rjordens/work/nist/artiq/artiq/test/coredevice/rtio.py", line 214, in test_loopback_count
    self.execute(LoopbackCount, npulses=npulses)
  File "/home/rjordens/work/nist/artiq/artiq/test/hardware_testbench.py", line 68, in execute
    raise error from None
  File "/home/rjordens/work/nist/artiq/artiq/test/hardware_testbench.py", line 63, in execute
    exp.run()
  File "/home/rjordens/work/nist/artiq/artiq/language/core.py", line 197, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rjordens/work/nist/artiq/artiq/coredevice/core.py", line 96, in run
    object_map, kernel_library, symbolizer = self.compile(function, args, kwargs, set_result)
  File "/home/rjordens/work/nist/artiq/artiq/coredevice/core.py", line 88, in compile
    raise CompileError(error.diagnostic) from error
artiq.coredevice.core.CompileError:
<artiq>/test/coredevice/rtio.py:118: error: for statement cannot be interleaved because trip count is indeterminate
                for i in range(self.npulses):
                         ^^^^^^^^^^^^^^^^^^^
<artiq>/test/coredevice/rtio.py:118: note: this expression is not supported
                for i in range(self.npulses):
                               ^^^^^^^^^^^^
```

jordens, 2016-01-18:
@whitequark can we rewrite this to fit or do we need the alternative, transaction-style `with parallel`?

whitequark, 2016-01-18:
@jordens we need the latter.

sbourdeauducq, 2016-01-19:
I think we should be able to keep ``with parallel``, see https://github.com/m-labs/artiq/issues/193#issuecomment-172892165.

whitequark, 2016-02-16:
This is an extremely annoying issue that, on reflection, I'm unsure if I can handle robustly. In short, there are two propositions that the compiler has to prove:
  1. That the value of `self` is known at compile (i.e. interleaving) time;
  2. That the value of `self.npulses` does not change at runtime.

Unlike #192/#193, where the propositions are easy to prove and the challenge lies in introducing the semantics in a way that creates a great user experience, here we have a much more complex situation. The point (2) is impossible to prove without whole-program analysis and severe dataflow restrictions. It would not even be enough to completely inline all invocations that depend on attribute values into the toplevel function, because `self` can be aliased and changed elsewhere, in places where dataflow is still opaque.

Essentially, the only thing that would work reliably (in light of mutable attributes) is completely linearizing the program not only with regards to control flow but also data flow; making sure that in every case that an attribute is retrieved, the object from which the attribute is retrieved is always the same. Of course, this is far too restrictive, although this is not unlike how the old compiler worked.

Yet again, what I really need here is immutable fields, just like #193 begs for immutable bindings. If only there was a good way to express this in Python...

whitequark, 2016-02-16:
The point (1) can be expressed using the same mechanisms as used for #193 and it should be as powerful, so that part is less problematic.

sbourdeauducq, 2016-02-17:
This should not be needed as ``with parallel`` will use the "dumb" implementation that should support this case. The interleaving implementation will be renamed ``with sequential``.

closed 2016-02-17 by sbourdeauducq