m-labs/artiq #237: RTIOUnderflow not caught in kernel

jordens, 2016-01-19:
```python
    @kernel
    def run(self):
        dt = seconds_to_mu(1000*ns)
        while True:
            try:
                for i in range(1000):
                    self.ttl_out.pulse_mu(dt)
                    delay_mu(dt)
            except RTIOUnderflow:
                dt += 1
                self.core.break_realtime()
            else:
                self.set_dataset("pulse_rate", mu_to_seconds(2*dt))
                break
```

```python
======================================================================
ERROR: test_pulse_rate (artiq.test.coredevice.rtio.CoredeviceTest)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/rjordens/work/nist/artiq/artiq/test/coredevice/rtio.py", line 206, in test_pulse_rate
    self.execute(PulseRate)
  File "/home/rjordens/work/nist/artiq/artiq/test/hardware_testbench.py", line 63, in execute
    exp.run()
  File "/home/rjordens/work/nist/artiq/artiq/language/core.py", line 197, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rjordens/work/nist/artiq/artiq/coredevice/core.py", line 105, in run
    self.comm.serve(object_map, symbolizer)
  File "/home/rjordens/work/nist/artiq/artiq/coredevice/comm_generic.py", line 525, in serve
    self._serve_exception(object_map, symbolizer)
  File "/home/rjordens/work/nist/artiq/artiq/coredevice/comm_generic.py", line 517, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.RTIOUnderflow: RTIO underflow at 329980808272 mu, channel 2, slack -6600 mu
```

The exception should have been caught in the kernel.

whitequark, 2016-01-19:
Unable to reproduce: this works as expected.

```python
from artiq import *
from artiq.coredevice.exceptions import *

class TestCounts(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl_out")

    @kernel
    def run(self):
        dt = seconds_to_mu(1000*ns)
        while True:
            try:
                for i in range(1000):
                    self.ttl_out.pulse_mu(dt)
                    delay_mu(dt)
            except RTIOUnderflow:
                dt += 1
                self.core.break_realtime()
            else:
                self.set_dataset("pulse_rate", mu_to_seconds(2*dt))
                break
```

jordens, 2016-01-19:
This is one of the unittests. Could you try
`ARTIQ_ROOT=. python setup.py test -s artiq.test.coredevice.rtio.CoredeviceTest.test_pulse_rate`
assuming that you have a useful `device_db.pyon` in `.`?

whitequark, 2016-01-19:
Yes, I just did that too and the test passes.

jordens, 2016-01-19:
This is pipistrellog here. Maybe the first guess (`1000 ns`) works for you on kc705 already. Could you try reducing that (a lot) so that it definitely underflows?

whitequark, 2016-01-19:
Ah yes.

whitequark, 2016-01-19:
Fixed now.

closed 2016-01-19 by whitequark