m-labs/artiq #241: rare PYON decoding failure of worker set_dataset

sbourdeauducq, 2016-01-20:
On Peter's machine, this code fails with ``ValueError: cannot create an OBJECT array from memory buffer`` on the ``set_dataset`` call when run by the master. The PYON unit tests pass, and decoding the string outside of the master also works. The problem is not present on my or Joe's machine.

```python
import time
import numpy as np
from artiq import *

class CrashPYON(EnvExperiment):
    def build(self):
        pass

    def run(self):
        def stripchart_roll(xs, x0):
            xsnp = np.roll(np.array(xs), -1)
            xsnp[-1] = float(x0)
            return xsnp
        current_bright_mean = self.get_dataset('detect_bright_mean',
                                               default=100)
        x = self.get_dataset("default_bright_mean_strip_x",
                             default=[current_bright_mean]*100)
        x = stripchart_roll(x, int(time.time()))
        self.set_dataset("default_bright_mean_strip_x", x,
                         broadcast=True, persist=True)
```

jordens, 2016-05-05:
@whitequark can you still reproduce this? I have seen that failure as well but in different context (complex numbers) and fixed it there. If yes, what is the failing pyon serialization?

whitequark, 2016-05-05:
Nope, doesn't fail anymore.

closed 2016-05-05 by whitequark

sbourdeauducq, 2016-05-06:
@jordens What was the fix? It's not obvious from your commits.

jordens, 2016-05-06:
I don't know.