m-labs/artiq #242: add floating point plotting support to GtkWave

sbourdeauducq, 2016-01-20:
The core device analyzer outputs RTIO slack as a 64-bit float expressed in seconds. It is nice to plot it with an analog view in GtkWave, but the latter only supports integer values there, resulting in a unwanted logarithmic-ish scale.

sbourdeauducq, 2016-01-23:
Example VCD file produced by the analyzer.
[foo.zip](https://github.com/m-labs/artiq/files/101701/foo.zip)

jordens, 2016-01-23:
For this specific case, why not just use 64 bit integer multiples of 1 ns? The useful granularity of the slack is on the order of 10 ns (cpu clock) and will be for the foreseeable future.

For frequencies and voltages the story might be different.

sbourdeauducq, 2016-01-23:
That's inconsistent with the way other numbers (e.g. DDS frequencies) are represented, plus this will be annoying when we have a board with a CPU frequency not a multiple of 1ns.

jordens, 2016-01-23:
Then this bug is not just about the slack.
And anyway: slack is not a precision quantity. It's useful resolution is more like 10 ns independent of the cpu speed.

dhslichter, 2016-01-25:
I think having slack in increments of machine units as a 64-bit integer would work, no?  "Logarithmic" scale is not that bad, this is a coarse debugging tool really.

sbourdeauducq, 2016-01-25:
I'd like to keep everything consistent and user-interpretable in the VCD: SI units and floating point.

sbourdeauducq, 2016-01-25:
Note that GtkWave can display floating point numeric values just fine, it just cannot plot them correctly.

dhslichter, 2016-01-25:
I am fine with logarithmic scale of slack for the plotting, since for my purposes the only times it will matter is when it gets small, and then I can look at the actual floating point value.  Seems like this would be more effort to fix properly than it's worth.

sbourdeauducq, 2016-03-31:
@jboulder Done. We now get the expected linear rise in RTIO slack.
[image]

sbourdeauducq, 2016-03-31:
Patch posted to the gtkwave-users ML.

sbourdeauducq, 2016-03-31:
(I do not have a link for it because Sourceforge being what it is, there seems to be some delay before the archive pages are updated)

sbourdeauducq, 2016-03-31:
[gtkwave_floating_point_plotting.diff.zip](https://github.com/m-labs/artiq/files/197171/gtkwave_floating_point_plotting.diff.zip)

closed 2016-03-31 by sbourdeauducq