m-labs/artiq #243: fix Qt bugs instead of working around them

sbourdeauducq, 2016-01-21:
https://github.com/m-labs/artiq/blob/dae63bd10cfb8e65d6d30921c5a719de39a05f36/artiq/gui/log.py#L230
https://github.com/m-labs/artiq/blob/dae63bd10cfb8e65d6d30921c5a719de39a05f36/artiq/gui/log.py#L239

sbourdeauducq, 2016-02-11:
92c0ede689dcd8d8393b06a86cc748428305a012

sbourdeauducq, 2016-02-19:
3b81dd5adc5c277ceefd18d802a7871350b72408

sbourdeauducq, 2016-02-21:
e3e42216a2037bf46a4f749ff8088a0507d783f9

sbourdeauducq, 2016-03-17:
633bbc457db7140cef2aa8eba4d78e40f248d450

HarryPoon000, 2025-07-09:
> [633bbc4](https://github.com/m-labs/artiq/commit/633bbc457db7140cef2aa8eba4d78e40f248d450)

workaround removed in bebd89c959a2c2df0fbe6a59b73a1c41f3c1749f

HarryPoon000, 2025-07-11:
> https://github.com/m-labs/artiq/blob/dae63bd10cfb8e65d6d30921c5a719de39a05f36/artiq/gui/log.py#L230

Workaround removed in bebd89c959a2c2df0fbe6a59b73a1c41f3c1749f
 
> https://github.com/m-labs/artiq/blob/dae63bd10cfb8e65d6d30921c5a719de39a05f36/artiq/gui/log.py#L239

No longer scrolls back to the top when rows are removed. Workaround may be removed. (Will open a PR)\
Tested by disabling the workaround and doing the following:

1. Adding a `remove_row` function to remove a selected row and connecting it to a context menu action, then repeatedly removing individual rows at various scroll positions
    ```py
    class LogDock(QDockWidgetCloseDetect):
      # ...snip... #
      def remove_row(self):
          idx = self.log.selectedIndexes()
          if idx:
              source_idx = self.proxy_model.mapToSource(idx[0])
              self.model.beginRemoveRows(QtCore.QModelIndex(), 
                                          source_idx.row(), 
                                          source_idx.row())
              self.model.removeRow(source_idx.row())
              del self.model.entries[source_idx.row()]
              self.model.endRemoveRows()
      # ...snip... #
    ```
2. Setting the `self.depth` attribute in `_Model` to a smaller number and sending multiple messages to the log to trigger the row removal in `timer_tick()`
    ```py
    class _Model(QtCore.QAbstractItemModel):
      def __init__(self, palette):
        # ...snip... #
        self.depth = 20
        # ...snip... #
      
      # ... #

      # NOTE: unchanged
      def timer_tick(self):
        # ...snip... #
        if len(self.entries) > self.depth:
            start = len(self.entries) - self.depth
            self.beginRemoveRows(QtCore.QModelIndex(), 0, start-1)
            self.entries = self.entries[start:]
            self.children_by_row = self.children_by_row[start:]
            for child in self.children_by_row:
                child.row -= start
            self.endRemoveRows()

      # ... #
    ```

HarryPoon000, 2025-07-15:
> [e3e4221](https://github.com/m-labs/artiq/commit/e3e42216a2037bf46a4f749ff8088a0507d783f9)

https://github.com/m-labs/artiq/blob/cb5727c1ee1c6f4a81f25ee62557c3957b1a622b/artiq/applets/simple.py#L271-L278

White garbage still showing when the window is not set to frameless.

[image]

HarryPoon000, 2025-07-15:
https://github.com/m-labs/artiq/blob/cb5727c1ee1c6f4a81f25ee62557c3957b1a622b/artiq/applets/simple.py#L271-L278

White garbage still showing when the window is not set to frameless

[image]

HarryPoon000, 2025-07-15:
> [3b81dd5](https://github.com/m-labs/artiq/commit/3b81dd5adc5c277ceefd18d802a7871350b72408)

(`frontend/artiq_gui.py` renamed to `frontend/artiq_dashboard.py` in 7453d85d5eba166335a21f096cc39583f0e1a725)

https://github.com/m-labs/artiq/blob/cb5727c1ee1c6f4a81f25ee62557c3957b1a622b/artiq/frontend/artiq_dashboard.py#L272-L276

Hack no longer required (will open a PR).

Tested by:
1. Opening a detached applet with some dataset to show on the window, then;
2. Disabling the workaround, then re-running `artiq_dashboard`

This was also tested with applets attached to the main window, which also embeds the contents as expected.

HarryPoon000, 2025-07-15:
#### Remaining workarounds if #2809 is merged

> [e3e4221](https://github.com/m-labs/artiq/commit/e3e42216a2037bf46a4f749ff8088a0507d783f9)

https://github.com/m-labs/artiq/blob/cb5727c1ee1c6f4a81f25ee62557c3957b1a622b/artiq/applets/simple.py#L279-L294

---

> [92c0ede](https://github.com/m-labs/artiq/commit/92c0ede689dcd8d8393b06a86cc748428305a012)

https://github.com/m-labs/artiq/blob/cb5727c1ee1c6f4a81f25ee62557c3957b1a622b/artiq/gui/applets.py#L395-L407

https://github.com/m-labs/artiq/blob/cb5727c1ee1c6f4a81f25ee62557c3957b1a622b/artiq/gui/applets.py#L492-L494