m-labs/artiq #245: coredevice crash when RPCing from within an `except:` block

jordens, 2016-01-26:
```python
from artiq.experiment import *

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        try:
            raise RTIOUnderflow()
        except RTIOUnderflow:
            self.core.break_realtime()
            self.do()

    def do(self):
        print("foo")
```

```python
(artiq)rjordens@688rj1:~/work/nist/artiq/run (master)$ artiq_run repository/test_raise.py
WARNING:artiq.coredevice.comm_generic:Mismatch between gateware (0.1+718.ga4dffaa) and software (0.1+744.g5444cd3) versions
Core Device Traceback (most recent call last):
  File "<artiq>/runtime/ksupport.c", line 229, in exception_handler
    vect, pc, ea);
artiq.coredevice.exceptions.InternalError(0): Hardware exception 6 at PC 0x42020584, EA 0x420149b0
Traceback (most recent call last):
  File "/home/rjordens/src/conda/envs/artiq/bin/artiq_run", line 9, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/rjordens/work/nist/artiq/artiq/frontend/artiq_run.py", line 151, in main
    return run(with_file=True)
  File "/home/rjordens/work/nist/artiq/artiq/frontend/artiq_run.py", line 137, in run
    raise exn
  File "/home/rjordens/work/nist/artiq/artiq/frontend/artiq_run.py", line 130, in run
    exp_inst.run()
  File "/home/rjordens/work/nist/artiq/artiq/language/core.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rjordens/work/nist/artiq/artiq/coredevice/core.py", line 107, in run
    self.comm.serve(object_map, symbolizer)
  File "/home/rjordens/work/nist/artiq/artiq/coredevice/comm_generic.py", line 524, in serve
    self._serve_exception(object_map, symbolizer)
  File "/home/rjordens/work/nist/artiq/artiq/coredevice/comm_generic.py", line 516, in _serve_exception
    raise python_exn
artiq.coredevice.exceptions.InternalError: Hardware exception 6 at PC 0x42020584, EA 0x420149b0
```

Removing either the `break_realtime()` or the `do()` makes it run.

whitequark, 2016-01-26:
Uh, what? It's an alignment exception at a 16-aligned address?

jordens, 2016-01-26:
still happens with current gateware/runtime (6383253)

whitequark, 2016-01-26:
This is almost certainly a problem in the compiler and not the runtime. I'll get to it soon.

whitequark, 2016-02-22:
OK, I figured out what triggers the issue. Building artiq_personality.c with `-DNDEBUG` causes it to skip all the printing, which seems to tickle some kind of bug somewhere. (Yeah, not very specific.)

whitequark, 2016-02-23:
Was a fairly obscure LLVM bug. https://github.com/openrisc/llvm-or1k/commit/ff2fe8c318eb7c934a2f2ac8da61a00d62becf1f

closed 2016-02-23 by whitequark