m-labs/artiq #249: Qt fails to use monospace font for log on Windows

sbourdeauducq, 2016-01-28:
The second column should use monospace font so that compiler diagnostics etc. are properly displayed.

Windows:
[image]

Linux:
[image]

whitequark, 2016-01-28:
A more appropriate title is "I fail to use the support for selecting a monospace font appropriate for the platform that Qt explicitly provides by `QFont::Monospace`, and instead use some `setFamily` call".

sbourdeauducq, 2016-01-28:
Doesn't seem to be that simple (http://stackoverflow.com/questions/1468022/how-to-specify-monospace-fonts-for-cross-platform-qt-applications) but we can try that.

whitequark, 2016-01-28:
Ah yes, it's `QFontDatabase` on Qt5.

sbourdeauducq, 2016-01-29:
Done in 1d92c08.

sbourdeauducq, 2016-02-14:
Tested OK on Windows.

closed 2016-02-14 by sbourdeauducq