m-labs/artiq #251: setup.py test versus subprocesses broken deeply

jordens, 2016-01-29:
1. @whitequark: why are `lit` and `OutputCheck` dependencies in `setup.py` but not anywhere in the conda packaging
2. @whitequark: `setup.py test` does not consider the levenshtein dependency fulfilled by the conda package and downloads/builds/uses its own in `.eggs` (there might be others like this)
3. @sbourdeauducq: `setup.py test` injects the local artiq package into `sys.path` but as soon as any of the tools starts a subprocess, its path is the original one. Thus, if the local artiq package is **not also** linked into the site python tree (`pip install -e .`), subprocesses will not see artiq. Also if **any other artiq** is actually linked or installed, they will find and use that (probably the reason for a lot of buildbot breakage). Also it uses the `.eggs/` installed `levenshtein` for the first process and the corresponding conda package for the others.

sbourdeauducq, 2016-01-29:
Do we need to run the tests via setup.py at all? Setuptools is messy and the least we touch it the better.

whitequark, 2016-01-29:
@jordens 

re 1: because conda does not have a separate category for test dependencies, these are not build dependencies (we do not run them during build) and I do not expect users of ARTIQ to run its internal tests

re 2: it seems like the pypi package is called "python-Levenshtein" which does not match what's installed in site-packages. This is irritating and I will look into it.

jordens, 2016-01-29:
@sbourdeauducq IMHO it is very nice to be able to run the unittests in a package without having to install it in any way
@whitequark then they can also be dropped from `setup.py`, otherwise they will be installed from pypi and used in `setup.py test`

whitequark, 2016-01-29:
@jordens That's kind of the point, e.g. I do not use conda for development but my system Python.

jordens, 2016-01-29:
Could'nt you then write the two dependencies as optional, like `lit[testing]`?

sbourdeauducq, 2016-01-29:
``PYTHONPATH=`pwd` python3.5 -m unittest artiq/test/*.py``

jordens, 2016-01-29:
Sure. Or use `nose` or `py.test` or `unittest discover`. But the conda builds and the setup.py need adjusting.

whitequark, 2016-01-29:
@jordens conda does not have optional dependencies

whitequark, 2016-01-29:
Also, we do not run builds or tests on ARTIQ via conda because it was broken somehow... I don't remember how, but I enthusiastically migrated it all to conda and then had to revert

sbourdeauducq, 2016-01-29:
@whitequark can you have the buildbot easily run the tests without ``setup.py``? Then I suggest we just remove all support for ``setup.py test``.

whitequark, 2016-01-29:
@sbourdeauducq sure, that's no problem at all. Can you also do it for migen and misoc then?

jordens, 2016-01-29:
@whitequark i am talking about setup.py. that has optional dependencies that would prevent OutputCheck or lit from being installed on `setup.py install`. Conda testing was probably broken in the way i described above.

sbourdeauducq, 2016-01-29:
@whitequark Sure.

whitequark, 2016-01-29:
OK so what's the new way of testing?

sbourdeauducq, 2016-01-29:
Get ``import artiq`` to do the right thing via ``PYTHONPATH`` or whatever, then ``python -m unittest artiq/test/*.py`` (or ``artiq.test`` or similar, if that works).

whitequark, 2016-01-29:
The latter doesn't work. The former as properly written (using `**/*.py`) will break on Windows but I guess it is fine as we can't build noarch packages on non-Windows either.

sbourdeauducq, 2016-01-29:
``cd artiq; cd test; python -m unittest *.py``

sbourdeauducq, 2016-01-31:
``python -m unittest *.py`` won't take into account subdirectories. ``python3.5 -m unittest discover`` will. We should just rename the tests to ``test_*.py`` so that it matches the default discover filename pattern.

sbourdeauducq, 2016-02-02:
See the ``newtesting`` branch.

jordens, 2016-02-14:
done.

closed 2016-02-14 by jordens