m-labs/artiq #252: use pandas (or pytables) to write hdf5 files

jordens, 2016-01-31:
We currently use `h5py` to save data. `h5py` is very low level and we miss out on a lot of higher level features that we would need to implement ourselves, like metadata and the inherent abscissa-ordinate association. Also `pandas` will be the tool that people will want to use to read and analyze that data and the missing metadata makes reading our results files with pandas really hard. The overhead is negligible.

sbourdeauducq, 2016-01-31:
The ARTIQ datasets also currently do not support such metadata. What features exactly should be added to result management and how?

jordens, 2016-01-31:
Primarily, one should be able to open the results with pandas. The easier solution there appears to be saving them with pandas. The associations can come later.

sbourdeauducq, 2016-01-31:
Are there pandas or h5py bugs that prevents them from being read right now?

sbourdeauducq, 2016-01-31:
Pandas seems to have lots of dependencies, including matplotlib, and is slow to load. No less than 405 modules are pulled in by ``import pandas`` on a fresh interpreter, as shown by:

```
>>> s1 = set(sys.modules.keys())
>>> import pandas
>>> s2 = set(sys.modules.keys())
>>> len(s2-s1)
```

sbourdeauducq, 2016-01-31:
By comparison, that number is 22 for artiq.

jordens, 2016-01-31:
Nonsense. You can't claim that artiq pulls in just 22 modules. `h5py` itself pulls in 222 modules! If you are worried about modules (they all are packaged and maintained in conda), then at least we should switch to `pytables` as the underlying `hdf5` interface.

And as I explained being unable to read `h5py` written files in `pandas` is not a bug in either. It is just a different level of abstraction.

sbourdeauducq, 2016-01-31:
I mean the basic ``import artiq`` pulls 22. Others pull more (e.g. ``artiq.experiments`` takes 184), which is still much less than pandas (those 405 modules are just for the top-level) and is way faster to import.

jordens, 2016-01-31:
`import artiq` needs 22 modules just to expose `__artiq_dir__` and `__version__`. Comparing that with `pandas` (or `h5py` for that matter) does not make any sense.

sbourdeauducq, 2016-01-31:
It does give an indication of how much a given package likes to carelessly load tons of modules.

jordens, 2016-05-05:
Let's revisit this once this materializes:
https://hdfgroup.org/wp/2015/09/python-hdf5-a-vision/
(also https://github.com/pytables/pytables/tree/h5py)
Also keeping #405 in mind.

jordens, 2016-06-18:
Fine for now.

closed 2016-06-18 by jordens