m-labs/artiq #254: Windows conda install: malformed version string

dhslichter, 2016-02-03:
Conda complains about a malformed version string when attempting to install artiq or artiq-kc705-nist_qc2.  Problem does not appear using conda in Linux.  

```
c:\Anaconda3>conda create -n artiq020216 artiq
Fetching package metadata: ........
Solving package specifications: .............An unexpected error has occurred, p
lease consider sending the
following traceback to the conda GitHub issue tracker at:

    https://github.com/conda/conda/issues

Include the output of the command 'conda info' in your report.

Traceback (most recent call last):
  File "C:\Anaconda3\Scripts\conda-script.py", line 4, in <module>
    sys.exit(main())
  File "C:\Anaconda3\lib\site-packages\conda\cli\main.py", line 195, in main
    args_func(args, p)
  File "C:\Anaconda3\lib\site-packages\conda\cli\main.py", line 202, in args_func
    args.func(args, p)
  File "C:\Anaconda3\lib\site-packages\conda\cli\main_create.py", line 49, in execute
    install.install(args, parser, 'create')
  File "C:\Anaconda3\lib\site-packages\conda\cli\install.py", line 333, in install
    update_deps=args.update_deps)
  File "C:\Anaconda3\lib\site-packages\conda\plan.py", line 411, in install_actions
    update_deps=update_deps):
  File "C:\Anaconda3\lib\site-packages\conda\resolve.py", line 1058, in solve
    installed, minimal_hint=minimal_hint, update_deps=update_deps)
  File "C:\Anaconda3\lib\site-packages\conda\resolve.py", line 808, in solve2
    dists = self.get_dists(specs, filtered=True)
  File "C:\Anaconda3\lib\site-packages\conda\resolve.py", line 718, in get_dists
    dists.update(self.all_deps(pkg.fn, max_only=max_only))
  File "C:\Anaconda3\lib\site-packages\conda\resolve.py", line 596, in all_deps
    add_dependents(root_fn, max_only=max_only)
  File "C:\Anaconda3\lib\site-packages\conda\resolve.py", line 576, in add_dependents
    for pkg2 in self.get_pkgs(ms, max_only=max_only):
  File "C:\Anaconda3\lib\site-packages\conda\resolve.py", line 544, in get_pkgs
    pkgs = [Package(fn, self.index[fn]) for fn in self.find_matches(ms)]
  File "C:\Anaconda3\lib\site-packages\conda\resolve.py", line 544, in <listcomp>
    pkgs = [Package(fn, self.index[fn]) for fn in self.find_matches(ms)]
  File "C:\Anaconda3\lib\site-packages\conda\resolve.py", line 360, in __init__
    self.norm_version = VersionOrder(self.version)
  File "C:\Anaconda3\lib\site-packages\conda\resolve.py", line 151, in __init__
    raise ValueError(message + "invalid character(s).")
ValueError: Malformed version string '0.9.10~a6d5e28': invalid character(s).
```

jordens, 2016-02-03:
#132

closed 2016-02-03 by jordens

sbourdeauducq, 2016-02-03:
@dhslichter You are pulling an outdated version of pyqtgraph that will not work with ARTIQ anyway. I guess it is coming from https://anaconda.org/fallen/pyqtgraph, please change your conda repository URLs to those in the ARTIQ documentation. The correct pyqtgraph version is 0.9.10-py_105+giteea3752.