m-labs/artiq #255: Unsatisfiable dependencies for conda install on Windows

dhslichter, 2016-02-03:
Installing conda packages on Windows -- some clauses restricting dependencies appear to be unsatisfiable.  Output is below:
```
C:\Users\rabi>conda create -n artiq-2016-02-03 artiq=0.1=py_798+gitb7de92e --alt
-hint
Using Anaconda Cloud api site https://api.anaconda.org
Fetching package metadata: ........
Solving package specifications: ................................
Error: Unsatisfiable package specifications.
Generating minimal hint:
[      COMPLETE      ]|##################################################| 100%

The following set of clauses is unsatisfiable:

not qt-4.8.7-vc14_4
not qt-4.8.7-vc10_4
not qt-4.8.7-vc9_5
artiq-0.1-py_798+gitb7de92e
not qt-4.8.7-vc10_5
artiq-0.1-py_798+gitb7de92e => quamash-0.5.1-py35_4 or quamash-0.5.1-py_5
not qt-4.8.7-vc9_4
quamash-0.5.1-py_5 => pyqt-4.11.4-py35_2 or pyqt-4.11.4-py35_3 or pyqt-4.11.4-py
35_4
quamash-0.5.1-py35_4 => pyqt-4.11.4-py35_2 or pyqt-4.11.4-py35_3 or pyqt-4.11.4-
py35_4
not qt-4.8.7-vc14_5
pyqt-4.11.4-py35_2 => qt-4.8.7-vc10_4 or qt-4.8.7-vc9_4 or qt-4.8.7-vc9_5 or qt-
4.8.7-vc14_5 or qt-4.8.7-vc10_5 or qt-4.8.7-vc14_4
pyqt-4.11.4-py35_3 => qt-4.8.7-vc10_4 or qt-4.8.7-vc9_4 or qt-4.8.7-vc9_5 or qt-
4.8.7-vc14_5 or qt-4.8.7-vc10_5 or qt-4.8.7-vc14_4
pyqt-4.11.4-py35_4 => qt-4.8.7-vc10_4 or qt-4.8.7-vc9_4 or qt-4.8.7-vc9_5 or qt-
4.8.7-vc14_5 or qt-4.8.7-vc10_5 or qt-4.8.7-vc14_4

```
Here is the info for the current conda install:
```
C:\Users\rabi>conda info
Using Anaconda Cloud api site https://api.anaconda.org
Current conda install:

             platform : win-32
        conda version : 3.19.1
  conda-build version : 1.19.0
       python version : 3.5.1.final.0
     requests version : 2.9.1
     root environment : C:\Anaconda3  (writable)
  default environment : C:\Anaconda3
     envs directories : C:\Anaconda3\envs
        package cache : C:\Anaconda3\pkgs
         channel URLs : http://conda.anaconda.org/m-labs/channel/main/win-32/
                        http://conda.anaconda.org/m-labs/channel/main/noarch/
                        http://conda.anaconda.org/m-labs/channel/dev/win-32/
                        http://conda.anaconda.org/m-labs/channel/dev/noarch/
                        https://repo.continuum.io/pkgs/free/win-32/
                        https://repo.continuum.io/pkgs/free/noarch/
                        https://repo.continuum.io/pkgs/pro/win-32/
                        https://repo.continuum.io/pkgs/pro/noarch/
          config file : C:\Users\rabi\.condarc
    is foreign system : False
```

Here is a list of the installed packages:

```

C:\Users\rabi>conda list
# packages in environment at C:\Anaconda3:
#
_license                  1.1                      py35_1
alabaster                 0.7.7                    py35_0
anaconda                  2.4.1               np110py35_0
[... 129 lines cut ...]
sockjs-tornado            1.0.1                    py35_0
sphinx                    1.3.5                    py35_0
sphinx-rtd-theme          0.1.9                     <pip>
sphinx_rtd_theme          0.1.9                    py35_0
spyder                    2.3.8                    py35_0
spyder-app                2.3.8                    py35_0
sqlalchemy                1.0.11                   py35_0
statsmodels               0.6.1               np110py35_0
sympy                     0.7.6.1                  py35_0
tables                    3.2.2                     <pip>
tk                        8.5.18                   vc14_0  [vc14]
toolz                     0.7.4                    py35_0
tornado                   4.3                      py35_0
traitlets                 4.1.0                    py35_0
ujson                     1.35                     py35_0
unicodecsv                0.14.1                   py35_0
werkzeug                  0.11.3                   py35_0
wheel                     0.26.0                   py35_1
xlrd                      0.9.4                    py35_0
xlsxwriter                0.8.4                    py35_0
xlwings                   0.6.4                    py35_1
xlwt                      1.0.0                    py35_0
zeromq                    4.1.3                    vc14_1  [vc14]
zlib                      1.2.8                    vc14_2  [vc14]
```

dhslichter, 2016-02-03:
This issue is fixed if one first creates an environment for python 3.5, and then installs artiq into that environment:

```
C:\users\rabi> conda create --name artiq-2016-02-03 python=3.5
C:\users\rabi> activate artiq-2016-02-03
[artiq-2016-02-03] C:\Users\rabi>conda install artiq=0.1=py_798+gitb7de92e
```

This was the fix before for issue #133 , but I am surprised because this is a new Anaconda install with Python 3.5 only.  This issue does not appear when doing a conda install on Linux.  Perhaps something with dependencies in the win32 version of the quamash package?

sbourdeauducq, 2016-02-04:
We are going to switch to Qt5, so whatever the pyqt/quamash packages are doing wrong should become irrelevant.

closed 2016-02-04 by sbourdeauducq