m-labs/artiq #257: release process / clean up labels on anaconda.org

sbourdeauducq, 2016-02-13:
Released/stable packages (e.g. dependencies) should go to the ``main`` label. Those on autobuilder (migen, misoc, artiq, pythonparser, m-labs/pyqtgraph) should go to the ``dev`` label unless they are a release. At the moment, the labels attached to packages are chaotic.

I suppose we can move autobuilder packages manually from ``dev`` to ``main`` when they correspond to a release (= git tag).

The idea is users of ARTIQ releases would only add our ``main`` channel. The documentation should be updated to reflect that, too.

jordens, 2016-02-18:
When we have this resolved, track the recommended procedure in https://github.com/m-labs/artiq/blob/master/RELEASING.rst

whitequark, 2016-02-22:
I corrected the problem with packages from conda-recipes ending up in the `dev` channel a while ago. Also, I have just went through the packages and fixed labels everywhere.

There isn't any particular recommended procedure in addition to what you already described in `RELEASING.rst`, I just went through the anaconda.org interface and manually altered every one. This is annoying and error-prone (the "Move" and "Remove" menu items are quite close and there's no confirmation on "Remove"... stupid) but no other way.

sbourdeauducq, 2016-02-22:
Let's keep this open until the docs are updated - right not they still tell to use the dev channel.

whitequark, 2016-02-22:
Ah yeah. I missed that part.

whitequark, 2016-03-09:
The docs were fixed in d0d56bd3.

closed 2016-03-09 by whitequark