m-labs/artiq #259: hostname for the buildserver

jordens, 2016-02-14:
The buildserver should get a human readable hostname.

whitequark, 2016-02-14:
Yes. There is a static IP that should be assigned to the server, but it's unclear whether the ISP has done that yet or not. This should be sorted out (it should've been sorted out months ago...)

jordens, 2016-02-14:
Even if it is a dynamic IP currently, the best estimate is that is stable enough to give it a NS record.

whitequark, 2016-02-14:
I think the issue was that the `m-labs.hk` DNS was very annoying to change because of some organizational issues. I can put it under `whitequark.org` DNS which can be changed with one click, though.

jordens, 2016-02-14:
That would be fine by me.

whitequark, 2016-02-14:
http://m-labs-buildserver.whitequark.org

closed 2016-02-14 by whitequark