m-labs/artiq #261: update qt/pyqt conda packages after Qt 5.6 is out

sbourdeauducq, 2016-02-15:
Qt 5.6 brings welcome improvements to the docking system:
https://woboq.com/blog/qdockwidget-changes-in-56.html

jordens, 2016-07-23:
c.f. #243

dtcallcock, 2016-08-04:
When Qt 5.6 is implemented will the grouped dragging described in the blog post above be enabled? It's apparently disabled by default.  This feature is highly desirable as it would make managing large numbers of graph applets easier.  Also, is this likely to be 2.x or 3.0?

sbourdeauducq, 2016-08-05:
I had tried enabling it (it's only one line of code) on my machine that has up-to-date Qt from Arch Linux and it works. I don't know how it would integrate with saving/restoring applet positions though.

sbourdeauducq, 2016-11-15:
Anaconda now has a Qt5 package partially based on ours and upgraded to 5.6. Need to have ARTIQ use that one and test that it works correctly on Windows and Linux.

jordens, 2017-05-12:
3c50c46a81958d0295e829bbea05499a35385603 and #670

closed 2017-05-12 by jordens