m-labs/artiq #262: support cancelling all experiments from a pipeline

sbourdeauducq, 2016-02-15:
I propose doing this by right clicking on a experiment in the scheduler, and having a "request graceful termination of all experiments in the same pipeline as the one selected" option. A short text in the menu could be "Gracefully terminate all in pipeline".

jordens, 2016-02-15:
Wouldn't this be more appropriately placed in the context menu in the scheduler? Terminating a complete pipeline only makes sense if there is something in that pipeline. That provides a nice context.

sbourdeauducq, 2016-02-15:
Yes, I was talking about the context menu of the scheduler here: "propose doing this by right clicking on a experiment **in the scheduler**"

jordens, 2016-02-15:
Ack. Nevermind then.

commit dc2c19f0 (2016-03-29): gui/schedule: support terminating all experiments in a pipeline. Closes #262
closed 2016-03-29 by sbourdeauducq