m-labs/artiq #269: applet bugs on Windows

sbourdeauducq, 2016-02-16:
- [ ] the contents aren't displayed until you resize the dock
- [ ] there are white borders on two sides of the applets
- [ ] windows intermittently fail to be embedded (seems to happen only when the applets are already present at GUI startup)

sbourdeauducq, 2016-02-17:
Screenshot of the "white borders" (which also contain grey and blue here) on @cr1901's machine. Also note the setGeometryDp errors in the console. On Linux, the "hello world" button fills the whole window.

[image]

See this:
http://stackoverflow.com/questions/32612627/embedding-a-native-window-in-qt

sbourdeauducq, 2016-02-18:
host.py/embedded.py
[embedding.zip](https://github.com/m-labs/artiq/files/135833/embedding.zip)

sbourdeauducq, 2016-02-18:
This is what you can get by restarting a docked applet. Could be that those issues are all related the the failing ``setGeometryDp`` ...
[image]

cr1901, 2016-02-18:
I'll test on my machine and see if I can duplicate within the next 24 hours.

Evil-Spirit, 2016-02-19:
i think it's because window can't be resized due to some reason (like WM_GETMINMAXINFO response). Ive tried to set hook(SetWindowsHookEx) on it, but something goes wrong way (i don't know why it isn't set for me). in other words, you can interrupt winproc and return other minmax info instead of app-specified for ability to resize this window. with the resizable windows i have get a good result.

Evil-Spirit, 2016-02-19:
example (doesn't work for me)

http://paste.ofcode.org/v4qexKaVJAtpPvA22vZggv

sbourdeauducq, 2016-02-19:
Thanks and I see how we could work around this bug. Can you modify embedded.py so that it shows the window before sending the embedding request? Qt should make the window resizable after that (at least, you can resize it manually).

On Linux we had to use this sequence of operations to work around problems:
https://github.com/m-labs/artiq/blob/92c0ede689dcd8d8393b06a86cc748428305a012/artiq/applets/simple.py#L133
but likely the sequence that works on Windows is show applet window and then embed it.

sbourdeauducq, 2016-02-19:
28c4d8c2a7ba4f59fa3582f26879970e82004c84 makes things better (contents are shown immediately and there are no more problems with restarting), but there are still issues. The white borders are still there, Qt keeps complaining about ``setGeometryDp``, and one occasion I resized the applet window and the contents were not resized.

The applets failing to be embedded on GUI startup is a race condition in ``pipe_ipc``, I'll fix it shortly.

Are you able to reproduce the "white border" problem @Evil-Spirit ?

Evil-Spirit, 2016-02-19:
now i see it when running embedded.py and capture it by my qt program.

sbourdeauducq, 2016-02-19:
Correction: __some__ of the applets failing to be embedded are due to a ``pipe_ipc`` bug, which comes __in addition__ to an embedding bug. Sigh.

Evil-Spirit, 2016-02-19:
https://bugreports.qt.io/browse/QTBUG-40320

sbourdeauducq, 2016-02-19:
The other (non-pipeipc related) "window fails to be embedded" bug happens, reproducibly, when the applet is supposed to be embedded in a ``QDockWidget`` that is detached at application start. 

Embedding is otherwise reliable when the ``QDockWidget`` is docked into the main window when the application starts.

sbourdeauducq, 2016-02-19:
@Evil-Spirit Yes, see my comment in the ARTIQ code: "Qt window embedding is ridiculously buggy".

sbourdeauducq, 2016-02-19:
All "window not embedded" bugs fixed.

sbourdeauducq, 2016-02-19:
I suspect the white borders could be vestiges of the normal title bar and window border.

sbourdeauducq, 2016-02-19:
Applets failing to exit is a pyqtgraph bug I had fixed a while ago, but the conda package was out of date. I have rebuilt the pyqtgraph package, but when installing it, it attempts do downgrade Qt to 4.x. @whitequark please remember to fix this.

sbourdeauducq, 2016-02-21:
All fixed.

closed 2016-02-21 by sbourdeauducq