m-labs/artiq #279: pyserial asyncio controllers

sbourdeauducq, 2016-02-18:


sbourdeauducq, 2016-02-28:
Experimental asyncio support for pyserial is here:
https://github.com/m-labs/pyserial

The file ``aio.py`` can be used independently from the rest of pyserial. I have sent a few things to pyserial, if by next week they have not reacted to them we will host ``aio.py`` somewhere else, possibly as a different project.

sbourdeauducq, 2016-03-10:
Time's up. https://github.com/m-labs/asyncserial

sbourdeauducq, 2016-03-25:
Implemented in the two serial controllers (tcube/novatech), untested. I will give the tcube devices a shot. @jboulder can you test your controller for the Novatech 409b after the changes I made (in the master branch)?

asyncserial is currently not working on Windows.

sbourdeauducq, 2016-03-29:
Tcube driver OK.

sbourdeauducq, 2016-04-07:
Windows problems fixed.

sbourdeauducq, 2016-04-07:
Just waiting for the novatech test.

sbourdeauducq, 2016-08-03:
@jboulder Can you give the new novatech controller a try?

jbqubit, 2016-08-04:
I've ordered Novatech but they've not yet arrived. Will test when they're
on hand.

sbourdeauducq, 2016-08-21:
@jbqubit Have they arrived?

sbourdeauducq, 2016-09-15:
Testing this is the last known issue before 2.0.

sbourdeauducq, 2016-09-23:
Fixed and working.

closed 2016-09-23 by sbourdeauducq