m-labs/artiq #281: Failed to Create the Environment in Linux

lanlangugugu, 2016-02-20:
Hi,

I successfully created the environment yesterday. But when I tried it today, it fails:

lanlangugugu@Aiden:~$ ENV=$(date +artiq-%Y-%m-%d); conda create -n $ENV artiq-pipistrello-nist_qc1;     echo "Created environment $ENV for ARTIQ"dfadsfasdf
Using Anaconda Cloud api site https://api.anaconda.org
Fetching package metadata: ........
Solving package specifications: ..........................
Error: Could not find some dependencies for artiq-pipistrello-nist_qc1: artiq 0.1 py_887+gitd30fe60, quamash, artiq 0.0, flterm, pydaqmx, artiq 0.1 py_910+git4940e48, artiq 0.0 py_2382+git72a3a12, artiq 0.1 py_729+git8befc6a, artiq 0.1 py_921+gitfef7250, artiq 0.0 py_2290+git502e570, artiq 0.0 py_2378+gitf2f1deb, artiq 0.1 py_795+gitcf465da, artiq 0.1 py_863+git6ee86d9, artiq 0.0 py_2370+git73b4aff, artiq 0.0 py_2360+gitd633c8e, artiq 0.1 py_728+git0079aa4, artiq 0.1 py_924+git3b81dd5, artiq 0.0 py_2367+git400999f, artiq 0.1 py_774+git022a1ff, artiq 0.1 py_908+git2ce3c08, artiq 0.0 py_2331+git9ed6b54, artiq 0.1 py_874+git055573a, artiq 0.0 py_2221+git4def561, artiq 0.1 py_789+gitf14fed5, artiq 0.1 py_906+gitf0e2599, artiq 0.1 py_783+git627221a, artiq 0.0 py_2270+git25188f0, artiq 0.1 py_909+gitc0bbd99, artiq 0.1 py_798+gitb7de92e, artiq 0.0 py_2404+git7998679, artiq 0.1 py_745+git3a73673, artiq 0.1 py_888+gitc63afae, artiq 0.1 py_787+git17ac64a, artiq 0.1 py_858+git1cfca74, artiq 0.1 py_885+gitdf16a1d, artiq 0.0 py_2377+git027d54c, artiq 0.1 py_900+git617c416, artiq 0.0 py_2381+gitcc45694, artiq 0.1 py_714+gitdae63bd, artiq 0.1 py_747+git6383253, artiq 0.0 py_2385+git4198601, artiq 0.1 py_918+git1891725, artiq 0.0 py_2407+gita9cf892, artiq 0.1 py_898+git1465fe6, artiq 0.1 py_746+git7a9864b, artiq 0.1 py_709+git4dd0693, artiq 0.0 py_2241+git64ad388, artiq 0.0 py_2406+git1b97131, artiq 0.1 py_788+git6889149, artiq 0.1 py_711+gitb3ba97e, artiq 0.1 py_785+git107e2fe, artiq 0.1 py_907+git6196aaf, artiq 0.0 py_2181+git0d90d88, artiq 0.0 py_2399+git63f7899, artiq 0.1 py_727+git2c847b3, artiq 0.0 py_2302+git50a463a, artiq 0.1 py_822+gitdc955d4, artiq 0.1 py_911+gitca3cced, artiq 0.0 py_2277+gite4d73c0, artiq 0.1 py_913+git6973a94, artiq 0.1 py_730+gitdcea678, artiq 0.1 py_920+git4bd5a55, artiq 0.1 py_718+gita4dffaa, artiq 0.1 py_857+git734fa59, artiq 0.1 py_901+git122ddc4

Did you mean one of these?

    artiq-pipistrello-nist_qc1

Did you mean one of these?

    quamash

Did you mean one of these?

    artiq

Did you mean one of these?

    aterm

You can search for this package on anaconda.org with

    anaconda search -t conda artiq 0.1 py_901+git122ddc4

 (and similarly for the other packages)
Created environment artiq-2016-02-19 for ARTIQdfadsfasdf

I saw that the files were updated today. Is that the issue?

whitequark, 2016-02-20:
Yes, there is currently no functional pyqt package.

lanlangugugu, 2016-02-20:
Thx. I'll try installing from the source first.

sbourdeauducq, 2016-02-22:
@lanlangugugu This should be fixed now.

closed 2016-02-22 by sbourdeauducq