m-labs/artiq #282: Ubuntu 15.10+/Debian jessie+ specific instructions?

sbourdeauducq, 2016-02-20:
https://m-labs.hk/artiq/manual/installing.html#ubuntu-15-10-debian-jessie-specific-instructions

Who wants to maintain those, i.e. test them at every major ARTIQ release? And remove any superfluous packages?

They are also certainly incomplete, as they e.g. they do not include Qt, and it is not explained when they are needed (it seems they are for building from source a minimal ARTIQ that can just run kernels, no GUI/controllers).

If there are no maintainers, I will remove them.

whitequark, 2016-02-20:
I use Debian (and abhor conda) so naturally I know what's needed... don't care about Ubuntu though.

whitequark, 2016-02-20:
The list is wildly wrong though.

dhslichter, 2016-02-21:
These are so out of date that we should probably just remove them for now.

jordens, 2016-02-21:
That section was already hilariously broken when it was first written and committed. Removing would be beneficial.

jboulder, 2016-02-21:
When I was regularly building ARTIQ from source I was updating this list
for Ubuntu 14.04 LTE. I don't know when it shifted to 15.x. As part of the
build-install instructions a list of dependencies for a specific version of
Linux is crucial -- build errors in the absence of proper dependencies can
be quite cryptic to the uninitiated.

I strongly advocate for Ubuntu for historical reasons and because it is
officially supported by the NIST Linux security scripts. Details such as
build dependencies are for ease of installation by physicists not ARTIQ
developers. Peter's loathing of Ubuntu is noted but not relevant unless
there is very strong technical grounds to shift to a different
distribution.

What is the motive for a shift from 14.04 LTE to 15.x? Agreed that in
anticipation of v1.0 somebody needs to be installing and building on a
clean install of the standard disto. This should be a relative novice, not
one of the m-labs.hk developers. This may be an application of one of SB's
kode kids or a physicist.

Joe

dhslichter, 2016-02-21:
I think what we have right now is not useful so it is less confusing to remove it for the time being.  We should replace this with full instructions for installing ARTIQ/Misoc/Migen/openocd/llvm/clang and all dependencies on a fresh install of Xubuntu 14.04 LTS, aimed at the novice/non-Linux-expert ARTIQ user.  This particular choice of OS is because it is a preferred one for NIST users because of iTAC policy.  If someone has strong preferences for a different Linux distro, I would imagine it means they are sufficiently proficient with Linux to be able to figure out how to install the dependencies etc on their particular distribution without our having to document it in fine detail for them.

closed 2016-02-22 by sbourdeauducq

whitequark, 2016-02-22:
> loathing of Ubuntu is noted

Oh, I don't have strong feelings about Ubuntu. It's just that extracting the list of needed Debian packages is trivial and would take me a minute whereas for Ubuntu it's more involved (needs a new VM... etc).

jboulder, 2016-02-22:
The choice of Xubuntu 14.04 is an artifact of early testing at NIST on
Virtual Machines. Now that everybody at NIST is using dedicated hardware,
Ubuntu 14.04 is best. NIST security scripts officially support Ubuntu
14.04. -Joe

commit 14683f29 (2016-02-22): doc: remove 'Ubuntu 15.10+/Debian jessie+ specific instructions'. Closes #282