m-labs/artiq #287: [gui] applets grow in size

jordens, 2016-02-22:
If the GUI is maximized, then closed, then restarted, a docked XY plot applet has grown in size vertically.

sbourdeauducq, 2016-02-29:
Does not happen on Windows.

jordens, 2016-03-23:
only happens to me if the applet is vertically adjacent to the explorer

jordens, 2016-05-13:
On state restore a big number applet sometimes stays small. This is in the browser on the right hand dock area.

[image]

sbourdeauducq, 2016-05-13:
This one is probably another Qt window embedding bug unrelated to this issue. I suppose that it grows to the correct size when you resize the dock?

jordens, 2016-05-13:
Correct.

Spaqin, 2024-07-31:
Big number widget doesn't stay small anymore. Haven't noticed the issue with the widget increasing in size either. Instead I found another bug: fullscreen -> close -> restore, widgets are the size they would be when fullscreen. Trying to move (not even resize) the window causes the screen to go fullscreen; seems the state is remembered but it doesn't go fullscreen on startup.

[image]

Spaqin, 2024-08-01:
Generally fullscreen restore seems to be an [issue with Qt](https://bugreports.qt.io/browse/QTBUG-118598), rather than being ARTIQ specific. Should this be closed, or kept open for future reference then?