m-labs/artiq #289: run CI unittests on the KC705

sbourdeauducq, 2016-02-23:
Seems needed as they are broken basically every time I run them.

whitequark, 2016-03-08:
Done. They're still broken since either there is no loopback or there is no JTAG.

closed 2016-03-08 by whitequark

sbourdeauducq, 2016-03-08:
I see many failing ones that are not due to JTAG.

whitequark, 2016-03-08:
No, they are due to JTAG, because the fix is in master but was not flashed yet. Except one test that I asked you about on IRC.

whitequark, 2016-03-08:
(I've deliberately not flashed it manually so that we have an indication of flashing by buildbot starting to work.)

sbourdeauducq, 2016-03-08:
lda, analyzer?

whitequark, 2016-03-08:
Ah yes those are due to the missing `--no-localhost-bind` option. Not sure if we should add that or update Python.

sbourdeauducq, 2016-03-08:
I would monkey-patch asyncio if python version == 3.5.1. Earlier versions are not supported and later ones should be fixed (patch is committed).

sbourdeauducq, 2016-03-08:
What about the analyzer test failing? It seems the buildbot could not connect to the analyzer port on the core device.