m-labs/artiq #291: test_attenuation failing

whitequark, 2016-02-24:
```
$ ARTIQ_ROOT=. python3.5 -m unittest artiq.test.test_lda.TestLda                   
Traceback (most recent call last):
  File "/home/whitequark/.local/bin/lda_controller", line 9, in <module>
    load_entry_point('artiq', 'console_scripts', 'lda_controller')()
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/frontend/lda_controller.py", line 39, in main
    bind_address_from_args(args), args.port)
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/protocols/pc_rpc.py", line 531, in simple_server_loop
    loop.run_until_complete(server.start(host, port))
  File "/usr/lib/python3.5/asyncio/base_events.py", line 373, in run_until_complete
    return future.result()
  File "/usr/lib/python3.5/asyncio/futures.py", line 274, in result
    raise self._exception
  File "/usr/lib/python3.5/asyncio/tasks.py", line 240, in _step
    result = coro.send(None)
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/protocols/asyncio_server.py", line 28, in start
    limit=4*1024*1024)
  File "/usr/lib/python3.5/asyncio/streams.py", line 116, in start_server
    return (yield from loop.create_server(factory, host, port, **kwds))
  File "/usr/lib/python3.5/asyncio/base_events.py", line 963, in create_server
    sock.listen(backlog)
OSError: [Errno 98] Address already in use
first connection attempt to ::1:3253[<class 'artiq.protocols.pc_rpc.AutoTarget'>] failed, retrying in the background
Traceback (most recent call last):
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/protocols/pc_rpc.py", line 284, in __init__
    self.__coninit(firstcon_timeout)
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/protocols/pc_rpc.py", line 300, in __coninit
    (self.__host, self.__port), timeout)
  File "/usr/lib/python3.5/socket.py", line 711, in create_connection
    raise err
  File "/usr/lib/python3.5/socket.py", line 702, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
EE
======================================================================
ERROR: test_attenuation (artiq.test.test_lda.TestLda)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/test/test_lda.py", line 12, in test_attenuation
    test_vector = [i*step*dB for i in range(0, int(attmax*int(1/step)+1))]
TypeError: unsupported operand type(s) for /: 'int' and 'NoneType'

======================================================================
ERROR: test_attenuation (artiq.test.test_lda.TestLda)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/test/hardware_testbench.py", line 34, in tearDown
    self.stop_controller(name)
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/test/hardware_testbench.py", line 58, in stop_controller
    client = Client(desc["host"], desc["port"], target_name, t)
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/protocols/pc_rpc.py", line 103, in __init__
    self.__socket = socket.create_connection((host, port), timeout)
  File "/usr/lib/python3.5/socket.py", line 711, in create_connection
    raise err
  File "/usr/lib/python3.5/socket.py", line 702, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

----------------------------------------------------------------------
Ran 1 test in 1.013s

FAILED (errors=2)
```
In `device_db.ddb`:
```
    "lda": {
        "type": "controller",
        "best_effort": true,
        "host": "::1",
        "port": 3253,
        "command": "lda_controller -p {port} --bind {bind} --simulation"
    },
```
The port 3253 is not bound.

jordens, 2016-02-24:
`--no-bind-localhost` for now. see the other issue.

whitequark, 2016-02-24:
Ah, #253.

closed 2016-02-24 by whitequark