m-labs/artiq #294: analyzer overflow on pipistrello

jordens, 2016-02-25:
This is on pipistrello, current bitstream, bios, runtime (0.1-982-g51e831c).
Analyzer FIFO before DRAM writer is 1024 entries (up from 128, no obvious difference in behavior).

```python
from artiq.experiment import *

class BlinkForever(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("ttl2")
        self.setattr_device("ttl3")

    @kernel
    def run(self):
        delay(10*us)
        for i in range(1):  # 1: looses messages, 10: garbles output
            with parallel:
                self.ttl2.pulse(100*ns)
                self.ttl3.pulse(100*ns)
```

GIves from coreanalyzer (DMA buffer cleared before):

```
WARNING:artiq.coredevice.analyzer:analyzer FIFO overflow occured, some messages have been lost
Log channel: 24
DDS channel: 23
DDS one-hot: False
OutputMessage(channel=4, timestamp=1213189863619, rtio_counter=1213187725640, address=1, data=0)
```

With 10 iterations:
```
WARNING:artiq.coredevice.analyzer:analyzer FIFO overflow occured, some messages have been lost
Log channel: 24
DDS channel: 23
DDS one-hot: False
InputMessage(channel=82, timestamp=4294967296, rtio_counter=94489280512, data=607642970786627584)
InputMessage(channel=82, timestamp=4294967296, rtio_counter=85899345920, data=608613702115000649)
InputMessage(channel=82, timestamp=4294967296, rtio_counter=85899345920, data=609144216475402569)
InputMessage(channel=82, timestamp=4294967296, rtio_counter=85899345920, data=609733245470245193)
InputMessage(channel=82, timestamp=4294967296, rtio_counter=85899345920, data=610259533582827849)
InputMessage(channel=82, timestamp=4294967296, rtio_counter=85899345920, data=610856156079849801)
InputMessage(channel=82, timestamp=4294967296, rtio_counter=85899345920, data=611406083692429641)
InputMessage(channel=82, timestamp=4294967296, rtio_counter=85899345920, data=611942714086261065)
InputMessage(channel=82, timestamp=4294967296, rtio_counter=85899345920, data=612483502008435017)
InputMessage(channel=82, timestamp=4294967296, rtio_counter=85899345920, data=613047001717670217)
```

Dumps of that are attached.

[analyzer.zip](https://github.com/m-labs/artiq/files/146355/analyzer.zip)

sbourdeauducq, 2016-03-15:
8a6873cab25faa22a3be89a3543845e80af3f4dd should fix this, though it is largely untested as the KC705 is not working atm.

jordens, 2016-03-15:
works.

closed 2016-03-15 by jordens