m-labs/artiq #296: Build the bitstream: ValueError: max() arg is an empty sequence

lanlangugugu, 2016-02-25:
Hi,

When I was building the bitstream, there is an issue:

(artiq-2016-02-25)lanlangugugu@Aiden:~/artiq-dev$ export PATH=/usr/local/llvm-or1k/bin:$PATH
(artiq-2016-02-25)lanlangugugu@Aiden:~/artiq-dev$ echo $PATH
/usr/local/llvm-or1k/bin:/home/lanlangugugu/anaconda3/envs/artiq-2016-02-25/bin:/usr/local/llvm-or1k/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games
(artiq-2016-02-25)lanlangugugu@Aiden:~/artiq-dev$ python3.5 -m artiq.gateware.targets.pipistrello
make: Entering directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libcompiler_rt'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libcompiler_rt'
make: Entering directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libbase'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libbase'
make: Entering directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/liballoc'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/liballoc'
make: Entering directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libm'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libm'
make: Entering directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libdyld'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libdyld'
make: Entering directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libunwind'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libunwind'
make: Entering directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libnet'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/libnet'
make: Entering directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/bios'
 CC       main.o
 LD       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m misoc.tools.mkmscimg bios.bin
make: Leaving directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/bios'
make: Entering directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/liblwip'
rm -f lwipopts.h
rm -f arch
ln -s /home/lanlangugugu/artiq-dev/artiq/artiq/runtime/liblwip/lwipopts.h lwipopts.h
ln -s /home/lanlangugugu/artiq-dev/artiq/artiq/runtime/liblwip/arch arch
mkdir -p core/ipv4
mkdir -p netif
mkdir -p netif/ppp
make: Leaving directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/liblwip'
make: Entering directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/runtime'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/lanlangugugu/artiq-dev/misoc_nist_qc1_pipistrello/software/runtime'
Traceback (most recent call last):
  File "/home/lanlangugugu/anaconda3/envs/artiq-2016-02-25/lib/python3.5/runpy.py", line 170, in _run_module_as_main
    "__main__", mod_spec)
  File "/home/lanlangugugu/anaconda3/envs/artiq-2016-02-25/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/lanlangugugu/artiq-dev/artiq/artiq/gateware/targets/pipistrello.py", line 233, in <module>
    main()
  File "/home/lanlangugugu/artiq-dev/artiq/artiq/gateware/targets/pipistrello.py", line 229, in main
    builder.build()
  File "/home/lanlangugugu/artiq-dev/misoc/misoc/integration/builder.py", line 142, in build
    run=self.compile_gateware, **kwargs)
  File "/home/lanlangugugu/artiq-dev/misoc/misoc/integration/soc_core.py", line 195, in build
    self.platform.build(self, *args, **kwargs)
  File "/home/lanlangugugu/artiq-dev/migen/migen/build/xilinx/platform.py", line 28, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/lanlangugugu/artiq-dev/migen/migen/build/xilinx/ise.py", line 190, in build
    self.map_opt, self.par_opt)
  File "/home/lanlangugugu/artiq-dev/migen/migen/build/xilinx/ise.py", line 97, in _run_ise
    settings = common.settings(ise_path, ver, "ISE_DS")
  File "/home/lanlangugugu/artiq-dev/migen/migen/build/xilinx/common.py", line 19, in settings
    ver = max(vers)
ValueError: max() arg is an empty sequence
(artiq-2016-02-25)lanlangugugu@Aiden:~/artiq-dev$ 

P.S, Maybe that's because I didn't installed Vivado properly?

Please look at it, thank you!

jordens, 2016-02-25:
Pipistrello is a Spartan 6. That needs ISE and doesn't work with Vivado. See the ARTIQ manual.

whitequark, 2016-02-25:
The error message should clearly state this.

jordens, 2016-02-25:
The manual already states it in great detail.

The rest is a `migen` issue. Feel free to reopen there.

closed 2016-02-25 by jordens