m-labs/artiq #299: use Xvnc with QEMU

sbourdeauducq, 2016-02-28:
Having to compile and install the hacky VNC client is inconvenient, and it lacks features like mouse motion events plus special keys get stuck at times.

Alternatively, fix the QEMU VNC bugs.

whitequark, 2016-02-28:
I installed tightvncserver on the machine. You can now connect to win7-experimental.lab.m-labs.hk using the same SSH reflection trick that you did until now, but with different target host (`ssh -L 5900:win7-experimental.lab.m-labs.hk:5900 lab.m-labs.hk`; this resolves the domain name on the remote host). Alternatively you can connect directly from the office.

whitequark, 2016-02-28:
This is better than Xvnc because you get clipboard, waste less bandwidth on updates e.g. of area under cursor, and in general requires no funny initscripts or something like that.

closed 2016-02-28 by whitequark

sbourdeauducq, 2016-02-29:
Thanks. Did you report the VNC bugs to QEMU, by the way?

whitequark, 2016-02-29:
No, I'm not motivated to triage them.