m-labs/artiq #304: support for bit-banged I2C buses

sbourdeauducq, 2016-03-01:


sbourdeauducq, 2016-03-06:
PCA9548 works, TCA6424A needs testing. @dhslichter can you try it?

dhslichter, 2016-03-07:
Will test as soon as possible.

dhslichter, 2016-03-22:
PCA9548 works, can switch between HPC and LPC I2C buses.  TCA6424A does not appear to work currently.  Error is that "ack" bit is not received, but it appears necessary 9th clock pulse is not being sent from I2C master.  @raghavendrasrinivas will send further details.

sbourdeauducq, 2016-03-27:
Code in ARTIQ is OK but hardware was not, need to know whether 799c2c77c4e561e455159e84994d82998f610bab was necessary.

sbourdeauducq, 2016-04-06:
@r-srinivas Can you test again with the TCA devboard and ARTIQ 1.0rc2 (which does *not* have 799c2c77c4e561e455159e84994d82998f610bab)?

r-srinivas, 2016-04-07:
I used 1.0 rc2 and it seems to work fine. I was running this script on the eval board,

```python
from artiq.experiment import *

class I2C_Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("pca")
        self.setattr_device("tca")

    @kernel
    def pulse(self, sequence):
        dt=1*ms
        if sequence == 0:
            self.tca.set(0x020000)
        elif sequence == 1:
            self.tca.set(0x010000)
        elif sequence == 2:
            self.tca.set(0x070000)
        elif sequence == 3:
            self.tca.set(0x0b0000)
        delay(dt)
    
    def run(self):
        self.pca.set(2)
        self.tca.set(0)
        self.tca.set(0x030000)

        for i in range(10):
            self.pulse(0)
            self.pulse(1)
            self.pulse(2)
            self.pulse(3)
```

And you can see the LEDs flashing sequentially. So I think we can close it for now. I'll bring it up again if we have new issues.

closed 2016-04-08 by sbourdeauducq