m-labs/artiq #308: report RTIO output event collisions and RT2WB deadline failures

jordens, 2016-03-01:
An RT output event can be wrongly timed (will not meet RT) either because it collides with a previous output event (the phy is busy; that should be reported: https://github.com/m-labs/artiq/blob/master/artiq/gateware/rtio/core.py#L216) or because the WB target takes longer than expected (for "real-time") to ack the wishbone transaction because of e.g. its downstream logic being busy (shared buses, busy machinery). For the SPI core this happens only on writes: If you write while one transfer is being shifted and another is already pending, the write will be accepted when the in-flight transfer completes. This is not a timeout or error in the WB sense (because it will be accepted eventually) but it violates RT-ness and should be detected.

The RT2WB interface sounds like the natural point where we might want to enforce a RT timeout/deadline for every transaction (would be two cycles for SPI) and then detect and report both this RTIOTimeout case. And also report the RT output event collision case.

Currently RT2WB deadline is enforced only for reads and at the runtime level. That should change accordingly.

If a WB cores itself reports timeouts and err, then that should also be passed up through RT2WB. But that seems exotic in practice.

(c.f. discussion in https://github.com/m-labs/artiq/commit/6c899e6ba64ff83d9fb14ccbe048e14650d0e7c9)

jordens, 2016-03-07:
C.f. #320 as well.
I would like to do the following: 

To me a collision on `set_o()` due to a too small timestamp delta w.r.t. SERDES granularity has the same semantic value as a RT2WB collision due to a too small timestamp delta w.r.t. SPI bus or DDS bus activity. Whether in one case it is due to the details of the SERDES and in the other because of bus transactions does not matter at this level. The meaning is "timing collision due to output PHY business". Same for `set_oe()`. These are all "collisions" at the RTIO PHYs. I would wire them all up in one logic path and one exception.

Remove the `Error` from `RTIOCollisionError`. The negative connotation is clear, like for `RTIOUnderflow` and `RTIOOverflow`. Alternatively add the `Error` the the later two.

@sbourdeauducq comments?

whitequark, 2016-03-07:
I think the cause of the collision should be clearly stated in the exception message. Collision of the type "not monotonically increasing timestamps" is different from "trying to write to two addresses with the same timestamp".

whitequark, 2016-03-07:
It also might be pretty neat if we put the kernel CPU PC into RTIO analyzer or something every time a transition is planned. Unwinding one level deep has no cost.

jordens, 2016-03-07:
`RTIOSequenceError` concerns itself with monotonicity. That's a different topic.

The underlying bug in #320 may be broken duplicate removal logic. But is should raise `RTIOCollision`.

The PHY logic runs rather independent of the CPU. I don't see how the PC counter (in addition or comparison to the slack) would help here. The three times when you submit the event that will cause an `RTIOCollision`, when it gets generated in the PHY, and when it gets read/emitted are in general all very different. The PC in at submission is relevant. Whether it is also relevant at generation and/or emission is questionable.

sbourdeauducq, 2016-03-08:
The logic for underflow/collision is very different from the one for overflow/busy.

The underflow/collision check is done in the system clock domain, and it is precise (the kernel-CPU immediately knows after submitting a command whether that particular command caused those errors).

The overflow/busy check is/would be in the RTIO clock domain, and it cannot be precise. I'm not sure how to best report a "busy" error to the kernel-CPU in a useful way.

jordens, 2016-03-08:
You ware right. I like detecting `RTIOCollision` (on TTL serdes channels) and `RTIOUnderflow` early as well because it correlates error-causing and error-receiving portions of the code. `RTIOOverflow` on input is only noticed late and raised on reading the channel. I don't think we can easily do better here. `RTIOBusy` (let's call it that to stress that it happened deep down at the PHY and is not predictable earlier) would be reported late and raised on submission of the next event. This is a bit awkward. But unless we go for signals/IRQs, is there a way around it?

commit cb6c2800 (2016-03-09): Merge branch 'rtiobusy' (closes #308)

* rtiobusy:
  exceptions: clarify RTIOBusy
  gateware/rtio: factor _BlindTransfer
  test_spi: break_realtime
  test_spi: simplify test, add collision vs busy test
  hardware_testbench: clean up artiq_core_exeption printing
  hardware_testbench: also print artiq_core_exeption
  tests: test spi business
  rtio: cleanup RTIOBusy message
  rtio: remove unused include
  spi: fix frequency_to_div()
  hardware_testbench: don't allow unused *args
  coredevice: add RTIOBusy to __all__
  rtio: add RTIOBusy
closed 2016-03-09 by jordens