m-labs/artiq #310: clearly report gateware timing failure in buildbot

sbourdeauducq, 2016-03-02:
Fail the build, or issue a warning. (Sometimes failed-timing bitstreams are still usable)

whitequark, 2016-03-02:
Is there a reliable way to detect one?

sbourdeauducq, 2016-03-02:
Grep ISE logs for "All constraints were met", which means timing passed. I'm not sure about Vivado, nor if there is a better way than grepping the ISE log. One would have to read the Xilinx docs and experiment.

jordens, 2016-03-02:
for vivado if the stdout contains a line that looks like `WARNING: [Route \d+-\d+] Router estimated timing not met.`. For ISE, `\d+ constraint not met.`.

whitequark, 2016-03-02:
Should I extract some block from stdout for ease of analysis? If yes, how is it delimited?

jordens, 2016-03-02:
You could look at the xml files vivado and ise leave behind. Maybe they are easier to parse.

whitequark, 2016-03-02:
Yes but is it helpful if I give you the xml output? (I have no idea how do people debug timing failures.)

sbourdeauducq, 2016-03-02:
Simply having timing pass/fail information on IRC and in the global build logs e.g. with colors (all similar to build pass/fail) would already be helpful.

whitequark, 2016-03-02:
Sure, but since the extraction techniques for pass/fail and some extended report info are nearly same, I can do the latter also.

sbourdeauducq, 2016-03-02:
The useful information would be the names of the top-3 signal with the worst negative slack in each failed clock domain.

jordens, 2016-03-02:
The xml is not helpful for a human. Just if you want to extract stuff it is sometimes easier than plain text. This is in top_timing.rpt (vivado) and top.twr (ise). But they look painful to extract. On ISE we have top.twx with the same info in xml. I think vivado can be made to generate that timing report in xml as well.

whitequark, 2016-03-02:
I don't need XML, if there is consistent stdout output, then it is far easier to extract. XML is a pain to parse anyway. I can grab top_timing.rpt from the builder if you want, sure.

sbourdeauducq, 2016-03-15:
Since this is lagging, let's keep it simple and grep the log as in https://github.com/m-labs/artiq/issues/310#issuecomment-191193856

sbourdeauducq, 2016-06-24:
Ping

whitequark, 2016-06-24:
Done

whitequark, 2016-06-24:
This fails the build, let me know if you'd rather get a warning.

closed 2016-06-24 by whitequark