m-labs/artiq #312: move ethernet timing constraints of kc705 to MiniSoC

enjoy-digital, 2016-03-02:
To be done when we'll figure out a reliable way to apply constraint on signals with Vivado ("dont_touch" signal attribute?)

sbourdeauducq, 2016-08-15:
What's the current status on that?

enjoy-digital, 2016-10-27:
There are two things:
- 1)"attribute keep of foo is true" is not recognized by Vivado and we have to use "(* dont_touch = “yes” *) wire foo" and add support for that in migen.build.
- 2)Timings constraints for ethernet should be moved to MiniSoC target in MiSoC.

We could do 2) without 1) but we still have to fix 1) since I'm not sure timings constraints are correcly applied when building with Vivado (the Vivado log should show it when applying the XDC constraints, is there a log somewhere?)

sbourdeauducq, 2016-10-27:
http://buildbot.m-labs.hk/builders/artiq-board/builds/134/steps/conda_build/logs/stdio

sbourdeauducq, 2016-10-27:
Could this be the reason for the Ethernet bugs we are seeing with some switches?

enjoy-digital, 2016-10-27:
Not sure but possible:

WARNING: [Vivado 12-507] No nets matched 'eth_clocks_rx'. [/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/work/misoc_nist_clock_kc705/gateware/top.xdc:1027]
CRITICAL WARNING: [Vivado 12-4739] create_clock:No valid object(s) found for '-objects [get_nets eth_clocks_rx]'. [/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/work/misoc_nist_clock_kc705/gateware/top.xdc:1027]

sbourdeauducq, 2016-10-30:
Done.

closed 2016-10-30 by sbourdeauducq