m-labs/artiq #313: compiler stumbles over lambdas

jordens, 2016-03-02:
Current git master

```python
from artiq.experiment import *

a = kernel(lambda b: b)

class Foo(EnvExperiment):
    def build(self):
        self.setattr_device("core")

    @kernel
    def run(self):
        return a(3)
```

```
(py35)rj@tito:~/work/nist/artiq/run (master)$ artiq_run repository/test_crash.py
Traceback (most recent call last):
  File "/home/rj/src/conda/envs/py35/bin/artiq_run", line 9, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 199, in main
    return run(with_file=True)
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 185, in run
    raise exn
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 178, in run
    exp_inst.run()
  File "/home/rj/work/nist/artiq/artiq/language/core.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rj/work/nist/artiq/artiq/coredevice/core.py", line 99, in run
    object_map, kernel_library, symbolizer = self.compile(function, args, kwargs, set_result)
  File "/home/rj/work/nist/artiq/artiq/coredevice/core.py", line 80, in compile
    stitcher.finalize()
  File "/home/rj/work/nist/artiq/artiq/compiler/embedding.py", line 490, in finalize
    inferencer.visit(self.typedtree)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in visit
    return [self.visit(elt) for elt in obj]
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 39, in <listcomp>
    return [self.visit(elt) for elt in obj]
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/home/rj/work/nist/artiq/artiq/compiler/transforms/inferencer.py", line 921, in visit_Assign
    self.generic_visit(node)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 27, in generic_visit
    self.visit(getattr(node, field_name))
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 41, in visit
    return self._visit_one(obj)
  File "/home/rj/src/conda/envs/py35/lib/python3.5/site-packages/pythonparser/algorithm.py", line 32, in _visit_one
    return getattr(self, visit_attr)(node)
  File "/home/rj/work/nist/artiq/artiq/compiler/transforms/inferencer.py", line 836, in visit_CallT
    return self.visit_builtin_call(node)
  File "/home/rj/work/nist/artiq/artiq/compiler/transforms/inferencer.py", line 817, in visit_builtin_call
    assert False
AssertionError
```

The thing is stumbles over is a `<function kernel>`.
If lambdas are a problem it should complain.

whitequark, 2016-03-03:
This is a bug in `inspect.getsource`. For your example it returns a meaningless result of `a = kernel(lambda b: b)`. And the worst part is that there's no reasonable way to extract a lambda from a line at all because the only information I have is `fn.__code__.co_firstlineno`, which is not enough to identify a particular lambda if there's more than one on a string. Typical Python garbage; it even has the column number in the AST but doesn't bother to record it.

whitequark, 2016-03-03:
Now displays this diagnostic:
```
test.py:11:9-11:13: fatal: lambdas cannot be used as kernel functions
        func()
        ^^^^  
test.py:3:1: note: lambda created here
func = kernel(lambda: None)
^                          
```

commit 73bfbe51 (2016-03-03): compiler: reject lambdas used as kernel functions (fixes #313).
closed 2016-03-03 by whitequark