m-labs/artiq #314: move SPIMaster to misoc

jordens, 2016-03-03:
The SPIMaster core should receive a CSR-based wrapper around SPIMachine in addition to the raw wishbone based wrapper. Verify that the core provides at least the functionality of the current SPI master in misoc. Add C header and api functions. Coordinate the change and alert people. Move to misoc.

The core could possibly be improved by making the transfer bit counters count the bits over all chained transfers and not per individual transfer. Then a chained transfer is complete and CS can be deasserted when the counters are zero, not when there is no pending transfer.

enjoy-digital, 2016-03-04:
I'm probably the only one using the SPI master in misoc. The actual SPI master was mainly 2 shift-registers (mosi/miso) controlled with CSR. Feel free to replace it with this new SPIMaster once the CSR-based wrapper will be implemented.

jordens, 2016-08-25:
https://github.com/m-labs/misoc/commit/504775c561fa93e60db1a3991abb33d851058642

jordens, 2016-08-25:
transitioning artiq needs new misoc release, package copying, doc update etc.

sbourdeauducq, 2016-08-26:
Is that for 2.0 or 3.0?

jordens, 2016-08-26:
i would do it now/before 2.0

commit 051e6e04 (2016-08-26): spi: use misoc SPIMachine, closes #314
commit fc4b3e44 (2016-08-26): spi: use misoc SPIMachine, closes #314
closed 2016-08-26 by jordens