m-labs/artiq #316: closes connection crashes with watchdog

jordens, 2016-03-03:
Current gateware, runtime, software versions.
I would expect a watchdog exception.

`artiq.test.coredevice.test_rtio.CoredeviceTest.test_watchdog` runs fine.
The comms cpu runs fine after.
No output on the console.

```python
from artiq.experiment import *

class Test(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("led")

    @kernel
    def run(self):
        self.core.break_realtime()
        self.led.on()
        with watchdog(.1*s):
            delay(.2*s)
            self.led.off()
            self.led.sync()
```

```
WARNING:artiq.coredevice.comm_generic:Mismatch between gateware (0.1+1045.g8adef12) and software (0.1+1078.g423ca03.dirty) versions
Traceback (most recent call last):
  File "/home/rj/src/conda/envs/py35/bin/artiq_run", line 9, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 199, in main
    return run(with_file=True)
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 185, in run
    raise exn
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_run.py", line 178, in run
    exp_inst.run()
  File "/home/rj/work/nist/artiq/artiq/language/core.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/rj/work/nist/artiq/artiq/coredevice/core.py", line 108, in run
    self.comm.serve(object_map, symbolizer)
  File "/home/rj/work/nist/artiq/artiq/coredevice/comm_generic.py", line 520, in serve
    self._read_header()
  File "/home/rj/work/nist/artiq/artiq/coredevice/comm_generic.py", line 101, in _read_header
    (sync_byte, ) = struct.unpack("B", self.read(1))
  File "/home/rj/work/nist/artiq/artiq/coredevice/comm_tcp.py", line 60, in read
    raise IOError("Connection closed")
OSError: Connection closed
```

sbourdeauducq, 2016-03-03:
There should be a "watchdog expired" entry in the core device log. This problem is merely that the watchdogs expiring currently close the connection without reporting a specific error code.

jordens, 2016-03-03:
IMHO watchdogs being un-catchable from within kernel code would be ok (if the alternative is difficult). But then I would like to catch them (the connection closing) from regular code.

sbourdeauducq, 2016-03-06:
Making them catchable should be doable by having the comms-CPU assert an IRQ line of the kernel-CPU. The problem though, it asynchronous exceptions in Python have race conditions (http://dept.cs.williams.edu/~freund/papers/python.pdf).

whitequark, 2016-03-06:
Speaking of race conditions, are you talking about ARTIQ Python or host Python?

sbourdeauducq, 2016-03-07:
The paper talks about a problem in host Python that we would reproduce if we implement asynchronous watchdog exceptions.

whitequark, 2016-03-07:
We won't have that problem in ARTIQ Python since ARTIQ Python code does not allocate or free resources (except memory but memory is inherently managed correctly in presence of stack unwinding).

sbourdeauducq, 2016-03-08:
The problem happens with all ``try...finally...`` blocks, no matter whether they allocate resources or not.

whitequark, 2016-03-08:
I don't think so. Would you write a specific example where this is a problem in ARTIQ Python?

sbourdeauducq, 2016-03-08:
There is another (independent) detail: what to do when a core device watchdog kicks in while waiting for a RPC to complete.

sbourdeauducq, 2016-03-08:
We should probably unittest this corner case too (not done atm).

sbourdeauducq, 2016-03-10:
Can we move the whole thing to 2.0?

jordens, 2016-03-10:
FWIW we can move the "catching watchdog timeouts on the core device" to 2.0. But if it is at all possible, I would like to have the band aid in 1.0 that I mentioned above: where I can catch the watchdog exception in the host-python layer that called the kernel. That seems completely unaffected by the other problems.

commit 501de306 (2016-03-18): Report watchdog expiration and RTIO clock failure as exceptions.

Fixes #316.
closed 2016-03-18 by whitequark