m-labs/artiq #317: reentrant kernels should be possible or raise errors

jordens, 2016-03-03:
```python
    @kernel
    def run(self):
        self.rpc()

    def rpc(self):
        self.run()
```

hangs.

I think this is detectable and should be an error.

jordens, 2016-03-03:
It does *not* hang because of the infinite recursion.

whitequark, 2016-03-09:
This *is* detected. This is what I get:
```
Core Device Traceback (most recent call last):
  File "test.py", line 9, in artiq_run_test.Test.run (RA=0x458)
    self.rpc()
  File "test.py", line 12, in rpc
    self.run()
builtins.OSError(5): Incorrect reply from device: _D2HMsgType.LOAD_FAILED (expected _D2HMsgType.LOAD_COMPLETED)
Traceback (most recent call last):
  File "/home/whitequark/.local/bin/artiq_run", line 9, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_run')()
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 199, in main
    return run(with_file=True)
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 185, in run
    raise exn
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/frontend/artiq_run.py", line 178, in run
    exp_inst.run()
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/language/core.py", line 192, in run_on_core
    return getattr(self, arg).run(run_on_core, ((self,) + k_args), k_kwargs)
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/coredevice/core.py", line 114, in run
    self.comm.serve(object_map, symbolizer)
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/coredevice/comm_generic.py", line 524, in serve
    self._serve_exception(object_map, symbolizer)
  File "/home/whitequark/Work/artiq-dev/artiq/artiq/coredevice/comm_generic.py", line 516, in _serve_exception
    raise python_exn
OSError: Incorrect reply from device: _D2HMsgType.LOAD_FAILED (expected _D2HMsgType.LOAD_COMPLETED)
```
and in the core log:
```
Attempted to load new kernel library while already running
```

closed 2016-03-09 by whitequark