m-labs/artiq #320: TTLInOut is extremely programmer-hostile

whitequark, 2016-03-07:
If you don't insert a delay of at least 8 mu between `ttlN.output()` and `ttlN.on()` then the output driver doesn't turn on but no error of any kind is produced. This shouldn't happen.

whitequark, 2016-03-07:
This isn't mentioned even in the documentation (neither in "Getting started" nor in the reference), though I think that a mere notice in the manual is hardly any good.

jordens, 2016-03-07:
You should get an `RTIOCollisionError`. `set_o()` and `set_oe()` being zero delay to the timeline is ok.

whitequark, 2016-03-07:
I vaguely recall @sbourdeauducq mentioning something to the degree that there should be a delay between them, which (and also watching for hi-Z vs driven low by poking the output with my finger) is how I figured this out at all

jordens, 2016-03-07:
If you had gotten an error you would have figured it out just as well.

Adding a delay would make so much of the beauty and utility of the multiple-FIFO design unnecessary and unused. The RFC on the mailing list provides a physics rationale as well.

whitequark, 2016-03-07:
Yes. There unquestionably should be an error. I just provided context for how I debugged it.

dhslichter, 2016-03-08:
I noticed this issue a couple weeks ago and told @sbourdeauducq ; he said he would add to the documentation but I guess this hasn't happened yet.  However, @whitequark this is definitely something that should return an error in the compiler.

whitequark, 2016-03-08:
Not in the compiler but in the runtime most likely.

dhslichter, 2016-03-08:
That's fine, but at least something to let the programmer know that there is an issue.

sbourdeauducq, 2016-03-08:
How is that not in the documentation?
https://m-labs.hk/artiq/manual/core_drivers_reference.html#artiq.coredevice.ttl.TTLInOut.output
"There must be a delay of at least one RTIO clock cycle before any other command can be issued."

https://m-labs.hk/artiq/manual/getting_started_core.html#connecting-to-the-core-device
"If the led device is a bidirectional TTL (i.e. TTLInOut instead of TTLOut), you need to put it in output (driving) mode. Add the following at the beginning of run:
self.led.output()
**delay(0.1*us)**"

sbourdeauducq, 2016-03-08:
Also, I think that generally the ``output`` calls will go into the startup kernel and there will be little to none input/output mode switching in experiments, so the requirement for the delay should not be a regular problem.

As for the lack of an exception, this comes from the pulse collapsing code in the RTIO core (``ttl.off()`` immediately followed by ``ttl.on()`` is equivalent to only ``ttl.on()`` if there is no underflow). I guess it could check that the RTLink addresses match.

commit 2953b069 (2016-03-08): rtio: when rtlink addresses are different, issue collision not replace (fixes #320)
commit ef217f7f (2016-03-11): Merge branch 'master' into scanwidget

* master: (125 commits)
  rtio: when rtlink addresses are different, issue collision not replace (fixes #320)
  rtio: collision_error -> collision
  test/coredevice/time_keeps_running: start new session to prevent now_mu save/restore
  runtime: fix sloppy memory management in cache_put.
  test_loopback: bump RTT limit to 60ns.
  soc: use add_extra_software_packages, factor builder code
  doc/PCA9548: clarify channel selection
  doc: I2C/QC2
  test: I2C/PCA9548 unittest
  examples/device_db: add PCA9548
  runtime/i2c_read: fix MSB
  ad5360: add documentati [... cut]
closed 2016-03-08 by sbourdeauducq