m-labs/artiq #324: coreanalyzer broken

jordens, 2016-03-09:
I went ahead and fixed the other two bugs. There are probably more:

```
(py35)rj@tito:~/.../artiq/run (master)$ artiq_coreanalyzer -d foo.dump -w fooo.vcd -p
Log channel: 25
Traceback (most recent call last):
  File "/home/rj/src/conda/envs/py35/bin/artiq_coreanalyzer", line 9, in <module>
    load_entry_point('artiq', 'console_scripts', 'artiq_coreanalyzer')()
  File "/home/rj/work/nist/artiq/artiq/frontend/artiq_coreanalyzer.py", line 51, in main
    print("DDS channel:", decoded_dump.dds_channel)
AttributeError: 'DecodedDump' object has no attribute 'dds_channel'
(py35)rj@tito:~/.../artiq/run (master)$
```

whitequark, 2016-03-10:
FYI there's also the bug where you need an even total number of RTIO messages or some of them get dropped. We've discussed it and decided not to add a workaround in runtime before it's fixed in gateware because the workaround is quite ugly and tricky to get right.

sbourdeauducq, 2016-03-10:
What two other bugs?

sbourdeauducq, 2016-03-10:
And again: the messages are not dropped, they are not moved from the buffer at an appropriate time.

closed 2016-03-10 by sbourdeauducq

jordens, 2016-03-10:
See the commit log.

sbourdeauducq, 2016-03-10:
I already did, if I commented here that's because I didn't find analyzer-related commits.

jordens, 2016-03-10:
The other two were multi DDS bus related.

commit ea523c76 (2016-03-10): frontend/coreanalyzer: do not attempt to print obsolete decoded_dump attribute. Closes #324
commit bc920345 (2016-03-11): Merge branch 'master' into scanwidget

* master: (38 commits)
  hardware_testbench: better message when skipping
  test_spi: drain errors and be more strict on where we expect errors
  monkey-patch asyncio.proactor_events to handle ConnectionAbortedError on Windows. Closes #247
  test/rtio/Loopback: ensure loop_out is low before starting test
  test/rtio: raise exception when pulse is not received
  rtio: fix different address collision detection
  frontend/coreanalyzer: do not attempt to print obsolete decoded_dump attribute. Closes #324
  coredevice: put cache into separate file/device
  gui [... cut]