m-labs/artiq #326: gateware runtime API versions

jordens, 2016-03-10:
Once the 1.0 dust settles, instead of asking for exactly matching git hashes for gateware and software (both in packaging and in protocol negotiation), we may want to introduce an API version between the runtime and the gateware (and maybe one between the runtime and the rest of the world). It would need increasing if the csr or the memory layout or the behavior of the gateware changes. Looks like manually tracking it will be required.

But it would reduce the amount of waiting, gateware building, and flashing. The gateware-software hash mismatch warnings are excellent but they are not fatal (nor shoud they be) and ignoring them could become a habit.

C.f. #97

dhslichter, 2016-03-15:
This seems reasonable to me but will obviously require considerably more manual effort on the part of m-labs to ensure that software and gateware remain compatible.  I am not extremely worried about this, whatever way people decide.

sbourdeauducq, 2016-06-24:
Even if the current solution is pretty heavy-handed, it has the advantage of clearly marking user-modified ARTIQ gateware as such.

sbourdeauducq, 2016-06-24:
I propose we keep it.

jordens, 2016-06-24:
It could still mark user-modified gateware as such.

But ok. Not a real issue for end-users anyway.

closed 2016-06-24 by jordens