m-labs/artiq #330: scan widget crash when dragging sliders

sbourdeauducq, 2016-03-13:
```Traceback (most recent call last):
  File "/home/sb/artiq/artiq/gui/scanwidget.py", line 388, in handleStopMoved
    self.sigStopMoved.emit(self.rangeToReal(rangeVal))
  File "/home/sb/artiq/artiq/gui/scanwidget.py", line 362, in rangeToReal
    pixelVal = self.slider.rangeValueToPixelPos(val)
  File "/home/sb/artiq/artiq/gui/scanwidget.py", line 114, in rangeValueToPixelPos
    opt = QtWidgets.QStyleOptionSlider()
RecursionError: maximum recursion depth exceeded while calling a Python object
Aborted (core dumped)
```

May be helped/triggered by hi-DPI display. The crash happens most of the time and renders the scan widget unusable.

jordens, 2016-03-13:
I haven't seen this but I bet the signal loops are in there. Unless @cr1901 finds time to fix that today, I'll rewrite that.

sbourdeauducq, 2016-03-14:
Works now.

sbourdeauducq, 2016-03-14:
(The changes still need to be merged into ARTIQ though, and scrolling is still missing from the widget)

commit 1dc72635 (2016-03-14): scanwidget: changes as of 19a1d40 (closes #330)
closed 2016-03-14 by sbourdeauducq