m-labs/artiq #331: fix determination of uninitialized variables with exceptions

sbourdeauducq, 2016-03-15:
```
    @kernel
    def readback(self):
        i2c_init(self.busno)
        i2c_start(self.busno)
        #r = 0
        try:
            if not i2c_write(self.busno, self.address | 1):
                raise I2CError("PCA9548 failed to ack address")
            r = i2c_read(self.busno, False)
        finally:
            i2c_stop(self.busno)
        return r

<artiq>/coredevice/i2c.py:72:16-72:17: error: variable 'r' is not always initialized here
        return r
               ^
<artiq>/coredevice/i2c.py:71:13: note: variable is not initialized when control flows from this point
            i2c_stop(self.busno)
```

whitequark, 2016-03-15:
Ah, yes, I remembered why this happens.

The problem is that, to prevent massive code duplication, the contents of the `finally` block are only codegen'd once. Of course, this prevents any flow analysis that only takes into account one previous block (as opposed to some longer path) from being very precise.

We could solve this easily at the cost of duplicating the contents of `finally` for every `except` clause plus once for `try` body. Ideally, this deduplication is something that should happen at a later stage, maybe even in LLVM, but the practical considerations are such that optimization passes aren't smart enough to do this transformation. It goes something like...

```
try:
  foo()
  return 1
except A:
  return 2
except B:
  raise C
finally:
  bar()
```
to
```
try:
  foo()
  k = 1; goto fin
except A:
  k = 2; goto fin
except B:
  k = 3; goto fin
label fin:
  bar()
  if k == 1:
    return 1
  if k == 2:
    return 2
  if k == 3:
    raise C
```

and it's pretty hard to do automatically.

whitequark, 2016-03-15:
Generally I think there are a few things we can do.
  1. Just duplicate code.
  2. Teach the liveness analysis the semantics of `finalize` blocks.
  3. Generate a more straightforward IR for analysis but then codegen the optimized version.

sbourdeauducq, 2016-03-15:
Okay. If the situation is understood and there is no bug hidden under this, the current behavior is fine as the user code is easy to adapt.

commit 77d47c2f (2016-06-22): transforms.artiq_ir_generator: split out finally→reraise control flow.

This makes it accessible to introspect by local access validator,
making some previously rejected code valid.

Fixes #331.
closed 2016-06-22 by whitequark