m-labs/artiq #333: changing scan type crashes GUI

sbourdeauducq, 2016-03-15:
[artiq_gui.pyon.zip](https://github.com/m-labs/artiq/files/174161/artiq_gui.pyon.zip)
select "Random scan" and:
``` 
Traceback (most recent call last):
  File "/home/sb/artiq/artiq/gui/scanwidget.py", line 404, in resizeEvent
    self.viewRange()
  File "/home/sb/artiq/artiq/gui/scanwidget.py", line 330, in viewRange
    self.realStop - self.realStart)
TypeError: unsupported operand type(s) for -: 'float' and 'NoneType'
Aborted (core dumped)
```

sbourdeauducq, 2016-03-15:
Unfolding argument groups with a scan widget in them produces the same crash.

jordens, 2016-03-16:
fixed in 11bfe33
@sbourdeauducq could you give it a hidpi test?

closed 2016-03-16 by jordens

sbourdeauducq, 2016-03-17:
Works great!