m-labs/artiq #334: dds batch not properly terminated

sbourdeauducq, 2016-03-16:
This raises ``DDSBatchError``.

```python
from artiq.experiment import *

class PulseRate(EnvExperiment):
    def build(self):
        self.setattr_device("core")
        self.setattr_device("core_dds")
        self.setattr_device("dds0")
        self.setattr_device("dds1")

    @kernel
    def run(self):
        dt = seconds_to_mu(100*us)
        while True:
            try:
                delay(10*ms)
                for i in range(100):
                    with self.core_dds.batch:
                        self.dds0.set(100*MHz)
                        self.dds1.set(100*MHz)
                    delay_mu(dt)
            except RTIOUnderflow:
                dt += 100
                self.core.break_realtime()
            else:
                print(mu_to_seconds(dt))
                break
```

I guess the compiler fails to call ``__exit__`` of the context manager under certain conditions.

sbourdeauducq, 2016-03-16:
Also, please add this to the rtio unittests after it works.

whitequark, 2016-03-18:
What is `core_dds`?

whitequark, 2016-03-18:
Unable to reproduce. I had to add the following changes to `device_db.pyon`:
```
    "core_dds": {
        "type": "local",
        "module": "artiq.coredevice.dds",
        "class": "CoreDDS",
        "arguments": {"sysclk": 3e9}
    },
    "dds0": {
        "type": "local",
        "module": "artiq.coredevice.dds",
        "class": "AD9914",
        "arguments": {"bus_channel": 0, "channel": 0},
        "comment": "Comments work in DDS panel as well"
    },
    "dds1": {
        "type": "local",
        "module": "artiq.coredevice.dds",
        "class": "AD9914",
        "arguments": {"bus_channel": 0, "channel": 1}
    },
```
The code prints `9.9999e-05` every time.

sbourdeauducq, 2016-03-18:
Fails here. Try reducing the initial value of ``dt`` so that you get some underflows.

whitequark, 2016-03-18:
Still works with `dt = seconds_to_mu(1*ns)`.

sbourdeauducq, 2016-03-18:
Out of curiosity, what does it print then?

whitequark, 2016-03-18:
`1e-09`

sbourdeauducq, 2016-03-18:
That's obviously broken in another way, you definitely cannot program DDSes in 1ns.

whitequark, 2016-03-18:
Well the bottom line is I still can't reproduce it. So, give me your complete environment, or something.

sbourdeauducq, 2016-03-18:
Fix this new bug first, then I guess you should be able to reproduce it.

whitequark, 2016-03-18:
Which new bug?

sbourdeauducq, 2016-03-18:
Underflow not raised (or not caught) when dt=1ns.

whitequark, 2016-03-18:
It is not "new". There have been no changes in compiler semantics at all since you filed this issue. It's some kind of misconfiguration, see my `device_db.pyon` above.

sbourdeauducq, 2016-03-18:
Your device db is identical to mine (I actually use the exact one which is committed), and it's new in the sense that I had not seen it before.

sbourdeauducq, 2016-03-18:
Also I cannot imagine any sort of misconfiguration that won't make the DDS error out at 1ns programming rate. There must be some serious bug in the code.

whitequark, 2016-03-18:
No, it wasn't... like I said it was misconfiguration. `"arguments": {"bus_channel": 0, "channel": 1}` instead of `"bus_channel": 26`.

whitequark, 2016-03-18:
In this case __exit__ is not called because the exception is raised from __enter__:
```
  File "<artiq>/coredevice/dds.py", line 38, in artiq.coredevice.dds._BatchContextManager.__enter__ (RA=0x1004)
    self.core_dds.dds_batch_enter()
  File "<artiq>/coredevice/dds.py", line 65, in artiq.coredevice.dds.CoreDDS.dds_batch_enter (inlined)
    dds_batch_enter(now_mu())
  File "/var/lib/buildbot/slaves/debian-stretch-amd64-2/miniconda/conda-bld/work/artiq/runtime/dds.c", line 182, in dds_batch_enter
    <unknown>
```

whitequark, 2016-03-18:
Oh, that's the wrong exception.

sbourdeauducq, 2016-03-18:
Hm. Even when attempting to push DDS commands on a TTLOut PHY, I'd have expected ``SequenceError`` or ``Collision`` at dt=1ns.

commit dbc0a899 (2016-03-18): dds.c: turn off batch mode before an underflow can be raised.

Fixes #334.
closed 2016-03-18 by whitequark