m-labs/artiq #336: name manual deployments from CI after their branch

jordens, 2016-03-18:
... otherwise master CI builds override release doc builds etc.

sbourdeauducq, 2016-03-19:
What about having the master CI upload docs in ``artiq/manual-dev`` and the releases in ``artiq/manual``? I don't think we need to keep all old manuals online.

whitequark, 2016-03-19:
Why not?

sbourdeauducq, 2016-03-19:
loads of files laying around, more complex folder structure. How would you name the paths?

whitequark, 2016-03-19:
manual-$BRANCH

sbourdeauducq, 2016-03-19:
And symlink manual to the latest?

whitequark, 2016-03-19:
No symlinking. That would break any existing links or bookmarks every time we release.

whitequark, 2016-03-19:
Instead, redirect.

sbourdeauducq, 2016-03-19:
That needs to be a deep redirect then. How do you do that?

whitequark, 2016-03-19:
Doesn't need to be. The manual itself uses relative links and something outside just links to `/manual`, which should appropriately redirect. Afterwards, all URLs will already point where they should.

sbourdeauducq, 2016-03-19:
There are already a few links pointing inside ``/manual``, but OK.

whitequark, 2016-03-19:
Yes, in that case a deep redirect is needed.

whitequark, 2016-03-19:
It's trivial to do with any reasonable webserver, e.g. Nginx or Apache, by using their rewrite module.

sbourdeauducq, 2016-03-19:
But I do not have access to the server configuration.

whitequark, 2016-03-19:
Then get a better server. This is one of the simplest things that any host is supposed to support.

Changing the URL underneath will lead to link rot and confusion. I would know; I did this mistake on http://rxr.whitequark.org and now dozens of inbound links from stackoverflow etc point to either 404 or, even worse, subtly wrong results. This is unacceptable.

jordens, 2016-03-20:
Currently it goes to `https://m-labs.hk/artiq/manual-%25(branch)s/index.html`. No variable replacement.

jordens, 2016-03-20:
I put a redirect 301 into htaccess.

closed 2016-03-20 by jordens