m-labs/artiq #339: pipistrello bit file location not found by artiq_flash

jboulder, 2016-03-20:
Following Installing ARTIQ instructions...
``$ conda install artiq-pipistrello-nist_qc1  ``
Conda successfully installs the bit file to the location specified in [build.sh](https://github.com/m-labs/artiq/blob/master/conda/artiq-pipistrello-nist_qc1/build.sh). 

```
$ ls ~/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq/binaries/pipistrello-qc1/
bios.bin  bscan_spi_xc6slx45.bit  runtime.fbi  top.bit
```
But this path is unknown to artiq_flash.
``` 
(pip2)britton@boshi:~/github/m-labs$ artiq_flash -t pipistrello
proxy gateware bitstream bscan_spi_xc6slx45.bit not found 
```
The following is needed but not documented.
```
$ artiq_flash -t pipistrello -d ~/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq/binaries/pipistrello-qc1
```

jordens, 2016-03-20:
That artiq directory is not the one where your artiq_flash lives. You have multiple artiq's installed.

closed 2016-03-20 by jordens

jboulder, 2016-03-20:
Please reread my bug report.The conda script does not install .bit into a location automatically found in artiq_flash. I know how to make things work. The point of this report to point out that at present new users will encounter errors if they follow the conda installation instructions verbatim. Have you tried to reproduce what I see? 
```
$ which artiq_flash
/home/britton/miniconda3/envs/pip2/bin/artiq_flash
```

jordens, 2016-03-20:
Is `python -c 'import artiq; print(artiq.__artiq_dir__)'` == `~/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq/`?

jboulder, 2016-03-20:
```
$ python -c 'import artiq; print(artiq.__artiq_dir__)'
/home/britton/miniconda3/envs/pip2/lib/python3.5/site-packages/artiq
```

jboulder, 2016-03-20:
I see the source of the confusion. "pip2" is the name of the conda environment I setup for pipistrello v2.

jordens, 2016-03-20:
Next time, please don't assert that you know what the bug is. This generally doesn't help.

jordens, 2016-03-20:
If you get the bitstreams from conda but the artiq package from a locally installed (`pip`/`setup.py` or `--user` oder `--develop`) version, you have to play several tricks to make the "easy access" work.

commit 8e41d501 (2016-03-20): doc: add pipistrello adapter explicitly (closes: #339)